Ultimaker/Cura #2774: Prime tower wasted too much material and time

651851996, 2017-11-13:
Can not let it in the non-switching layer like support？

651851996, 2017-11-13:
http://tu-img-1.aixinxi.net/o_1buqmvuvi1m36pku2ib2ibgc8a.jpg（Image upload failed orz）

651851996, 2017-11-13:
[image]
Image upload failed orz

ChrisTerBeke, 2017-11-14:
There's a few settings to adjust the prime tower. I'm sure that this is not the result of the default settings, so something must have been adjusted already there.

651851996, 2017-11-16:
Well, using multiple heads for more material requires more switching volume，but the default 20x20 is also very wasteful

Ghostkeeper, 2017-11-25:
The tower is hollow. You can precisely set the volume it needs to extrude per layer, so you can let it waste as much or as little material as you like. The default is 10 cubic mm, which comes down to a wall of 1.34mm thick with 0.1mm layer height and 20mm tower width.

The prime tower has to switch extruder every layer for its functionality. Normally you'd need one prime tower for each extruder since you want to prime the extruder before it starts printing on that layer. However Cura will make sure to start the next layer with the extruder it ended the previous layer on, so you need to switch less often. Then you only need 1 tower (for dual extrusion; for triple extrusion you'd need 2).

651851996, 2017-11-26:
My machine is a 2in1 design, so I need a Prime Tower of at least 20mm2, but Prime is also solid without switching. Hope that when there is no switch like support

Ghostkeeper, 2017-12-01:
If you need at least 20mm^2, then multiply that with your layer height and enter that as the prime tower prime volume. E.g. 20mm^2 * 0.1mm = 2mm^3. If you enter that you will get a hollow prime tower that extrudes the required amount of material.

Is that not what you meant? Perhaps you can make us a drawing to show exactly what you meant? It's kind of hard to understand.

closed 2018-03-02 by ChrisTerBeke