Ultimaker/Cura #2776: cura generate a wrong gcode.

ledinger, 2017-11-13:
When I insert the stl file flat to cura, cura generate a gcode with printtime 4 Minutes.
When I insert the same file upright, cura generate a gcode with printtime more then 2 hours.

Best regards

leo E aus S

ChrisTerBeke, 2017-11-13:
Can you give us some more information like the STL file and screenshots of how you slice it?

ledinger, 2017-11-13:
![cura 3 04 fehler](https://user-images.githubusercontent.com/33639181/32749261-1d10ed64-c8bf-11e7-80c4-f5cb0643
[image]
574c.png)
how can I sent you the png with this info? How can I sent you the stl file?

ChrisTerBeke, 2017-11-13:
Can you check the G-gcode preview for both to see how it sliced? 

* STLs you can upload by dragging and dropping them in the text field.

ledinger, 2017-11-13:
The stl file is not supported, I rename the stl to png and put the file to the text field.
[image]

rename the png to stl please.

ledinger, 2017-11-13:
I am a german and my english is not so good. I think for this I make some mistakes. Please excue me.

ledinger, 2017-11-13:
In my cura version 2.62 and 3.04 I can not open the gcode files.

ChrisTerBeke, 2017-11-13:
Haha downloading and renaming actually worked, magic...

Imported the model and it seems all faces are inverted, so it's only printing the holes and a brim:

[image]

You can fix some of this by enabling "Keep disconnected faces" in the custom settings panel, but ideally you want to get a proper STL file with the faces pointing outwards.

[image]

Closing this issue now as it's now a Cura issue.

closed 2017-11-13 by ChrisTerBeke

ledinger, 2017-11-13:
Thankyou verry much for your Help.

Best regards 

Leo E aus S.