Ultimaker/Cura #2778: a fatal exception has occured that we could not recover from!

divyangp689, 2017-11-14:
Version: 2.3.0
Platform: Windows-7-6.1.7601-SP1
Qt: 5.5.1
PyQt: 5.5.1

Exception:
Traceback (most recent call last):
  File "cura_app.py", line 61, in <module>
  File "D:\2.3\build\inst\lib\python3.4\site-packages\UM\Application.py", line 308, in getInstance
  File "D:\2.3\build\inst\lib\python3.4\site-packages\cura\CuraApplication.py", line 173, in __init__
  File "D:\2.3\build\inst\lib\python3.4\site-packages\UM\Qt\QtApplication.py", line 94, in __init__
  File "D:\2.3\build\inst\lib\python3.4\site-packages\UM\VersionUpgradeManager.py", line 107, in upgrade
  File "D:\2.3\build\inst\lib\python3.4\site-packages\UM\VersionUpgradeManager.py", line 281, in _upgradeFile
MemoryError

ChrisTerBeke, 2017-11-14:
Hi. Version 2.3.0 is quite old, please try the newer 3.0.x version and see if that works for you.

Ghostkeeper, 2017-11-20:
This was fixed in version 2.4: https://github.com/Ultimaker/Uranium/commit/6bea040f6b9584d94bd2d4acd87645be1ae05cdf

closed 2017-11-20 by Ghostkeeper