Ultimaker/Cura #2779: Improvement: Edit rotation

thesec, 2017-11-14:
I'd like to edit the rotation angle per axis.

ChrisTerBeke, 2017-11-14:
You mean an input field?

thesec, 2017-11-14:
Yep 👍

ChrisTerBeke, 2017-11-14:
Fair enough. Not sure why those don't exist yet indeed...

awhiemstra, 2017-11-15:
Because a given rotation can be represented in multiple forms. The most common reasonably user-friendly representation for these kinds of inputs is Euler angles. But several combinations of Euler angles can represent the same rotation when represented as a quaternion or a matrix. This can lead to strange effects where suddenly your inputs flip around or not matching with the values you typed in.

In short, it does not exist yet because it is actually a pretty hard problem.

thesec, 2017-11-15:
... already solved by other slicers and every CAD software you would not want the "world's most advanced 3D printing software" to stand behind ;D

Ghostkeeper, 2017-11-25:
Let's do some competitor analysis then.

Other applications:
* Slic3r has menu items for "Rotate around X", "Rotate around Y", "Rotate around Z". This brings up a pop-up where a user can input one angle, then press OK to apply.
* Simplify3D has a transformation dialogue where users can apply X, Y and Z rotation at the same time. This applies X first, then Y, then Z, which gives some weird results if you modify Z first and then X, because the X rotation is now in the Y direction. That's the problem that awhiemstra has in mind if I understood the spirit behind his post correctly.
* Blender has a hotkey R where you can type "R90" to rotate 90 degrees around the axis determined by the current camera position in relation to the model. The "90" is not in any visible text box. This is why people hate Blender.
* CloudCompare allows the user to enter an axis to rotate around and an angle. It also has an Apply button.
* AutoCAD allows the user to enter a number while rotating just like Blender, but they have a text box that appears next to the cursor while you're rotating so that you can actually see what you're typing.

I'd say that my preference goes for a solution similar to AutoCAD where the user can enter a number while the rotation is going on. But since we require the user to hold the mouse button while rotating it could be a bit more difficult to use for Cura users.

thesec, 2017-11-25:
Great overview, thanks @Ghostkeeper .

Blender has the most counterintuitive approach.

Simplest would be three text boxes X, Y and Z for entry in each axis and the angles applied as each box is confirmed explicitly by button or enter-key. Seems pretty WYSIWYG and straightforward as you do have the order only, not a matrix with each step taken.

Ghostkeeper, 2017-12-01:
That's the Simplify3D approach, yes. Like I said though, that also has some disadvantages that awhiemstra and I pointed out.

DDDirk, 2017-12-03:
In Blender you can see the angle at the bottom left. Or in the Properties Editor you can see the fields for each axis (if you don't see that, the visibility can be toggled with N).
If you just use R, indeed that rotates from the viewpoint, which can rotate the model around all three axes at the same time, as can be seen in the Properties Editor. But you can also specify the axis after the R, eg RX40 (also indicated at the bottom left).

All this can be done without ever touching the mouse. This is why I love Blender (well, one of the reasons). (An important reason some people hate Blender is propably that the steep learning curve that makes them miss elementary stuff like this.)

thesec, 2017-12-03:
This is not about blender, nor about keyboard shortcuts.

DDDirk, 2017-12-03:
Inspiration was looked for elsewhere, and the Blender methods work very fast. Having the grid in the background can also be very useful.
The usage of keyboard shortcuts is also very relevant if you have RSI. If you've never had that you will probably not understand the immense gravity of the problem.

thesec, 2017-12-03:
Blender was already mentioned. Keyboard shortcuts remain OT, add them separately.

DDDirk, 2017-12-06:
Yeah, but the info about Blender was wrong, so I corrected it.

And it's the most informative presentation, giving both the total angle from the viewpoint and the separate orthogonal angles. (Update: to see the former you might have to press T.)
However, having this info next to the object in a floating box might indeed be handier. But being able to permanently see the cumulative rotation is also handy. For example, in Blender if you first press RX10 and then RX5, at the bottom left you see the current rotation (5°) and in the top right (yes, too far apart) the cumulative rotation (15°).
How many text boxes can one want? :)

By OT do you mean On of Off Topic? :)
I'd say On, unless it's about adopting shortkeys across the board. Without that, with every separate item, the input method should also be addressed. And I am an extremely strong supporter of using as many shortkeys as possible. RSI can really ruin one's life (once you have it it's a constant struggle for the rest of your life).

thesec, 2017-12-06:
So you have RSI for entering coordinates only then? **hrhr**
Make it another, separate thread.

herrdeh, 2018-06-25:
Just my 2cts:
Consistency matters!
Therefore, the "rotate" control should work like the "move" control:

    Open the panel when clicking - with 3 boxes for numbers,
    Do the action {rotation) after number has been punched and "tab" key has
    been hit or the input box is left otherwise.

Thus, the commands will be observed sequentially and no mess will occur. The numerical control will not do anything else than the graphical input does at the moment. 

And the user does not have to get familiar with different behaviours in moving and rotating.

Ghostkeeper, 2018-07-04:
Our project manager removed this from our planning. We won't be getting time to implement this any time soon.

ianpaschal, 2018-07-31:
We just discussed this one as a team. Indeed, as Arjen said, when mucking about with matrices things get complicated fast (we actually have some other issues related to that... halp).

But, so long as the rotations are always relative and applied sequentially, this is no more likely to produce weird behavior than using the rotate tool. This is essentially the Simplify3D method, as seen in Ruben's analysis.

ianpaschal, 2018-07-31:
Devs, CURA-5609

fieldOfView, 2018-07-31:
Note that the Cura Lulzbot Edition has an implementation of "just three euler angles to represent the current rotation", and it is just about as buggy as one would expect.

Perhaps the trimesh module could be used to get a more reliable transformation tree and matrix decomposition (https://github.com/Ultimaker/Cura/issues/2875)

ianpaschal, 2018-07-31:
The key here is not to try and represent the current rotation. That's the really hard part. One thing on my to do list today is to look at what libraries we can use to help us not reinvent the wheel (badly) when it comes to matrix decomposition.

But for this issue it should be possible to avoid that entirely by making the inputs relative (always reset to zero and apply as soon as the box loses focus, the latter of which is also the default behavior for other transform tools).

fieldOfView, 2018-07-31:
But that would make the rotation boxes work differently than the move and scale boxes. In my opinion that would be quite bad, unless you make them also look/act differently than the aforementioned tools. So you could for example have three sets of "rotate by +/-90 degrees" buttons, but not fields. As soon as you have fields people expect them to behave "absolute" like the other fields.

ianpaschal, 2018-07-31:
That's my concern as well but it's something we discussed this morning as a team and felt we wanted to go ahead and implement it anyway.

We also felt something like a ±90º button sort of defeated the purpose since the whole reason we want this is to lay something at 22.5º or whatever. Ok, not the whole reason, but enough new utility to justify adding new UI elements.

Luke is on vacation right now but we plan to build a basic implementation and confer with him when he's back how we can differentiate its behavior from absolute transforms. One that comes to mind is how it's done in 3dsMax: have a toggle to switch between absolute or relative type-in mode, perhaps this could exist for position, rotation, and scale, but simply be greyed out on "relative" for rotate mode so that it's clear it's functioning as intended, and that at least for now it's not possible to type in the absolute rotation values.

We'll see about that in the future. As I said, I'm reading through trimesh today as well as some other resources we pulled up.

I'm also looking at how this is handled in the very robust Three.js library which I have a ton of experience with. [Thank god for very readable code with lots of whitespace.](
https://github.com/mrdoob/three.js/blob/dev/src/math/Matrix4.js)

fieldOfView, 2018-07-31:
> Thank god for very readable code with lots of whitespace.

...which contains no documentation on how to use it

ianpaschal, 2018-07-31:
Yeah, not really the point. I don't want to figure out what the Matrix4 class is for, I just want to see the math written out.

The prop/function `extractRotation` can, I suspect, be used to extract the rotation. 🤔 😛 

[Aha! Confirmed in the documentation!](https://threejs.org/docs/index.html#api/math/Matrix4)

Anarasha, 2018-12-13:
Now, I know this thread is old, but I had hoped this function existed so I tried to Google a way to add it. 
But if this is not viable currently, then perhaps a way to improve the current system is do what Meshmixer does - a graphically represented 5-degree increment snapping wheel that you aim for with your mouse so the rotation isn't that ruled by the object or how your camera is set currently. I constantly find myself overshooting by 30-40 degrees if my hand twitches and I'm not looking at the object from the side. I love Cura, but the lack of proper rotation control is beyond frustrating

FalconFour, 2019-02-26:
Take a look at XYZprinting's software, called (appropriately) XYZprint. It really kicks Cura's butt in a lot of ways - because of how simple it is, yet what fantastically beautiful prints come out of it. It has a side panel that manipulate X/Y/Z position, and X/Y/Z rotation, for each selected model. It's allowed me to do some fantastically convoluted stacking stunts in a print area, and done far more with my XYZ printer than I can get done with Cura and other printers.

The thing that makes its rotation controls simple is that they PERSIST per each model. So, when you rotate by 180 degrees along the Z axis (common flat-on-bed rotation), the software now simply marks that the model is rotated on Z 180 degrees. So the controls just say "Z: 180". You can only go 360 degrees, so all you need is a 0-360 rotation for each axis, and you can create any rotation combination that way.

I find the current rotation controls in Cura to be absolutely insane and infuriating - that they seem to apply "un-sticky" to the object, so when you rotate an object, it's fixed in stone - you can't see how much it's been rotated (e.g. when I accidentally rotate something Y175 instead of Y180 and it's now floating half-through the bottom of the board, half above the board, but I can't see that from the top-down view). It's also almost like a software mini-game side-quest to try manipulating the mouse cursor in a way that will produce the desired rotation, like a 180 degree flip on the Y axis to turn a part upside down (I print a lot of flat parts).

All you really need to do is persist the rotation figures per model in each of X/Y/Z figures, so one can say 0/0/0 to reset to normal, or 0/0/180 to flip it around on Z, maybe decide later you want it flipped upside down instead so you go 0/180/180, see it needs to be rotated 90 degrees and go 0/180/90... stuff like that. Really easy to do in nearly any other software... in Cura it can be this simple without a lot of headache, unless you over-think how people would want to rotate objects.

Ghostkeeper, 2019-02-27:
XYZPrint seems to have the same approach as Simplify3D print then (referring to [my earlier post](https://github.com/Ultimaker/Cura/issues/2779#issuecomment-346945766) in this thread), except that the current rotation angles are shown and transformations are absolute. Try rotating first the Z to 90 degrees, then rotating the X axis and see if it rotates around the X axis or around the Y axis. I can't try that right now because XYZPrint doesn't support Linux :(

But of course being able to view the current rotation is another desire that we could take a look at.

arielbernal, 2019-04-27:
Cura 2019! and we still don't know how to rotate a 3D object? I'm not asking for a quaternion representation, just euler angles to the relative position and a reset rotations button will suffice. Repetier has it and it works. Of course the order in which you apply the transformations matters, but it doesn't prevent you to apply basic rotations with precision.

nallath, 2019-04-29:
Well, we do. It's just that we're not able to explain why it's not that simple.  

But as always; Pull requests are welcome!

Ghostkeeper, 2019-04-29:
> Repetier has it and it works.

Oh good, because Repetier has the same behaviour: http://dulek.net/work/repetier_nonuniform_scale.mp4

npm12017, 2019-06-26:
the real question to be asked is, Why are you even considering 3d printing, if you can't figure out how to rotate an object by typing three numbers for each axis to have the object rotated by i.e. trial and error. 

When i used Simplified I changed the values accordingly until i was happy with the rotation of the object. In Cura, it is almost hair splitting just to do the same thing.

Blender does this well in three different ways:
By pressing the R button followed by the number of degrees be it - or + number. Freeroam rotate by double tapping the R button. Typing in the numbers in three different fields like in simplified3D.

The problem with the Gizmo, when I click on the axis I want to rotate by, its not a permanent click, and as soon as i turn the gizmo the thin line becomes unselected. 

It would be great if we could double click on the thin line, then highlights to rotate the mouse is then locked to it. Allowing the user to then double in empty space to exit the locked rotation mode.

leoheck, 2019-07-15:
This feature https://github.com/Ultimaker/Cura/issues/6013 duplicates this issue.

tablatronix, 2019-12-01:
Please add this, my computer lags so much (osx macbook) that the current UI is almost impossible to use.

FalconFour, 2020-01-07:
Lulzbot's edition of Cura does this now, as it turns out. It's running on some stanky old Cura base, too. Why can't Cura proper get it figured out?

[image]
(yes, a photo of a screen, but it's on a different system and I don't feel like Sneakernetting a screenshot ;) )

colinig, 2020-05-12:
[image]

[image]

Unclick snap rotation and u'll be able to rotate it degree by degree

Anarasha, 2020-05-12:
> 
> 
> [image]
> 
> [image]
> 
> Unclick snap rotation and u'll be able to rotate it degree by degree

I appreciate it, but what I meant with this was a textbox where you can type in how much to rotate it. This thread is 3 years old and I think everyone is still struggling with not being able to rotate with text input

tablatronix, 2020-05-12:
What @Anarasha this is a terrible UI choice and NOT at all accessible

thesec, 2020-05-12:
@tablatronix - the UI is not from @Anarasha but a quote from the previous comment. Making it or other accessible is off topic.

The initial request is now three years old and by now not only - imo - common across other slicers but also part of other Cura ports ... 

With the task just "recently" reopened and set to "in progress" I guess some waiting and a few more targeted drops might get us an accessible UI into the Cura mainline :)

tablatronix, 2020-05-12:
Sorry I typoed, I edited it, I meant to say "what they said", input boxes is what we are asking for not only snaps and mouse rotation.

Heikit69, 2020-05-17:
In my opinion 90°-rotate buttons in every direction would be great and help a lot

fieldOfView, 2020-05-25:
I have created a pull request that implements rotate-by-90-degrees buttons: 
https://github.com/Ultimaker/Uranium/pull/612

msn444, 2020-12-09:
The 90-degree rotate buttons are good, but they would be perfect if they were 45-degree buttons instead!

leoheck, 2020-12-09:
I would say 15 degrees.
--
Leandro Heck

Heikit69, 2020-12-09:
Please don't do this. I love the 90 degrees

leoheck, 2020-12-09:
90 and 45 are multiples of 15 dude.
So, I will suggest something. Keep it moving 15 because it is fine grain
- Hit and hold Ctrl while moving and it will move 45 degress.
- Hit and hold Shift and it will move 90 degrees
Simple, clean, good for many more cases.

Also, even better. An input field. With the default value as 15, but the user can type the angle he wants.
So, type the value. Let's say, 8 degrees... and the wheel will turn in steps of 8 degrees, customizable for any needs.

The same thing can be used for movement and scaling...

nallath, 2020-12-09:
The main reason for adding the 90deg options is that you use that way more often (Since not everyone uses the same convention of Z being up from the buildplate). So switching them to 45 degree isn't a really good option. It adds extra effort for how most people use it. 

The issue with type-ins is _what kind of_. Rotation in 3D space is messy (scale is too, but it's slightly more manageable). Rotating something with a set angle is super simple, but it does get tricky if you also want to display the current rotation. People do expect that if they type in a certain rotation (or first do rotation A and then Rotation B) that the numbers that they type in are what is displayed.

Ghostkeeper, 2020-12-18:
The rotation handles already snap to 15 degrees when dragging them. I think that's plenty of functionality to handle the 15 and 45 degree cases.

ysu2, 2021-11-28:
It's a real shame that this is missing; currently I'm unable to grab the blue (y-axis) rotational circle even after multiple restarts.
The input fields would go around this bug, otherwise I'm clueless what can I do - apart from transitioning to another slicer. :(

Claghorn, 2021-11-29:
OK, here is a solution that solves all possible objections: If I right click on one of the rotation circles, I get a dialog that pops up asking for the amount to rotate it. Since I can only click on one at a time none of the spurious objections about multiple rotations being specified can be relevant. Actually you could do the same thing over on the side in the rotation dialog with a pull down list where I could pick exactly one axis to type the rotation value, I wouldn't even have to learn that right click is possible then.

Ghostkeeper, 2021-12-08:
That's one of the solutions listed in one of the first replies in this topic, Claghorn: https://github.com/Ultimaker/Cura/issues/2779#issuecomment-346945766 . The solution that Slic3r had at the time.

BloodyRain2k, 2022-04-19:
Is this actually being considered to be implemented or has this fallen under the table again?

gaby64, 2022-12-13:
i want fractional degrees!

krumk, 2022-12-23:
One more vote for fractional degrees. Very valuable, especially when trying to achieve a consistent top layer.

TheShanMan, 2023-04-24:
It's disappointing to see something I've occasionally really wanted to make use of has been requested for _so long_ but gone nowhere. Occasionally I have a print so large that fitting it on the build plate requires placing it mostly diagonally but not exactly 45 degrees. I'd sure like to be able to type a precise angle, and then set my top layer lines to match it precisely.

I even tried right clicking on the rotation circle hoping I'd find it there, and then I read that very solution has been requested here.

C'mon, Cura folks. I get that this is an edge case but being an edge case means it could be implemented in a way that's not super evident such as right-clicking the rotation circle or having a rotate option on the model's right-click menu, rather than in a way that brings the sorts of complication that's been mentioned here.

truegret, 2023-05-23:
I am the creator/requestor of #15407, and as a creator/modeler/engineer, the solution is really quite simple and no need to complicate it by the use of Euler calcs (maybe it would be helpful for code development).  As I say, the solution would be very simple.

1) Select the model (in my case I want to align two models and have a a face on each model snap together (as I recall, obsolete Windows Autodesk 123D having this feature)
2) Select the Rotate tool
3) Deselect Snap Rotation
4) Select the desired axis of rotation
5) Have the "Degree" box be a Dialog Box
6) Enter the desired Angle
7) Enter <enter>

If another angle is desired for another axis, select a different axis for step 4 above and repeat.  Otherwise, allow for two faces (one on model A and one Model B and "Snap" them together.

It is NOT an edge, and is not difficult to implement.

FalconFour, 2024-05-22:
Oh man, there is no way we're still this far into Cura's maturity but we still don't have the basic ability to rotate by a specific dimension, really?

Definitely don't think this issue is going to die without a fight, this many times requested.

leoheck, 2024-05-22:
> We are cleaning our list of issues to improve our focus.

Cool. Clean the fucking list by implementing the basic features instead of just throwing requests into the trash bin.

truegret, 2024-05-22:
I am the creator/requestor of https://github.com/Ultimaker/Cura/issues/15407, and as a creator/modeler/engineer would still like the ability to enter a specific angle, please.

nallath, 2024-05-22:
> > We are cleaning our list of issues to improve our focus.
> 
> Cool. Clean the fucking list by implementing the basic features instead of just throwing requests into the trash bin.

You might not mean it like that, but can you calm down a bit? We setup the bot because there are a ton of issues that are no longer a thing, which is what it's supposed to prune. I don't love the solution either, but as you already stated; I'd love to just build things instead of having to deal with things that may or may not be an issue.

herrdeh, 2024-05-22:
I'd like to give a strong vote for this issue as well.
I think it's just a very basic feature, IMHO it's necessity does not really require any debate.

[22 more replies]