Ultimaker/Cura #2781: Feature: Optimize Print Volume

thesec, 2017-11-14:
https://www.michaelfogleman.com/pack3d/ documents an interesting approach of bin-packing the models for optimum print volume usage ... that would be an awesome feature to have, even "only" for a single layer of models.

thesec, 2017-11-14:
Have you read the "single layer"?
Am really wondering what makes you think that it is a horrible to make a more efficient use of the available print bed size on  X and Y!?

diegopradogesto, 2017-11-15:
I agree that this is very helpful for SLS, CJP, HP's MJF or even the Polyjet/Multijet as you don't need to think about supports and you can print the models in (almost) any position no matter the angle. In FDM, SLA or DLP this is not so useful.

thesec, 2017-11-15:
I completely disagree. All this is about is efficient use of the printing volume. Z is only a single dimension and in the sample used to illustrate the use for stacking models on top of each other ... 

Allowing models to interlock instead of just lying somehow adjacent next to another is useful for any 3D printer.

diegopradogesto, 2017-11-15:
Maybe packing in just one height is a good idea. Actually our colleague @jackha is working on a plugin that can pack multiple models in several "virtual build plates".
That one combined with packing and also the "Auto Orientation" plugin, could be a good way to place models automatically.

thesec, 2017-11-16:
Sculpteo is calling this nesting (sounds a proper term for it :) ) and has made it available in their freshly released fabpilot service; fyi.

Ghostkeeper, 2017-11-25:
Let's stick to the horizontal packing in any case.

We have convex hulls as collision areas because this greatly simplifies computation. That would be the first step to fix. Our current algorithm also has a grid upon which it places your models, which again simplifies computation because it makes it easy to perform collision checks really quickly.

One problem I see with this is that this naturally requires the application to change the rotation of the model. Even if you only rotate around the Z axis, people often orient their models for optimal quality, such as by aligning long lines with the diagonal for a Cartesian printer.

If you don't rotate and keep the grid and convex hull in, our current algorithm will come close to optimal (it's greedy and goes to pareto-optimal), because the problem becomes very easy. So I think that removing the convex hull would be the only real way to go forward. Even if you do small rotation steps, the grid will compress your output to almost randomness anyway so you'd need to get rid of the grid simplification, but it quickly becomes prohibitively expensive to compute then.

Removing the convex hull would require performing arbitrary polygon offsets in the front-end though. Quite a difficult problem!

stloewen, 2020-12-17:
What's the status of this? According to the release notes Cura v4.8 should be using Prusa's libnest2d which is based on SVGNest. However I'm still getting very different (and oftentimes very bad) results from Cura's auto-arrange compared to SVGNest or PrusaSlicer.

# Example
## Cura 4.8
[image]

## PrusaSlicer 2.2
[image]

nallath, 2020-12-21:
> What's the status of this? According to the release notes Cura v4.8 should be using Prusa's libnest2d which is based on SVGNest. However I'm still getting very different (and oftentimes very bad) results from Cura's auto-arrange compared to SVGNest or PrusaSlicer.

It is. If you have an issue with it, please create a new ticket for it (and include the files that reproduce the issue). Thanks!

closed 2020-12-21 by nallath