Ultimaker/Cura #2784: Request: Place center of right-click-rotation in iddle of part

Drayson80, 2017-11-14:
I experienced that the mouse click rotation is somewhere in the surounding area.
It shall be in the middle of the print plate or - if selected - in the center of the selected part.

ChrisTerBeke, 2017-11-14:
You can enable "Center camera when item is selected" to get this behaviour. It will focus on the last selected object on the build plate.

closed 2017-11-14 by ChrisTerBeke

Drayson80, 2017-11-15:
Thanks, already did...
I experienced a strange behavior (means lost center of rotation) when having all options selected (checked center camera..., invert driection..., zoom towards mouse...)
If I disable just "zoom towards mouse..." the behavior is as requested, so there it seems to be the reason for my request... could there be an conflicting interaction of those behaviors??

ChrisTerBeke, 2017-11-16:
That's by design as well since 'zoom towards mouse' will change the center of the camera target to the location of your mouse. When clicking on an object the camera will focus on the object again.

Ghostkeeper, 2017-11-25:
Also when you simply pan the view (shift + right-click drag) the origin moves. This is important because you can be zoomed in to a detailed part of your print and still want to rotate.