Ultimaker/Cura #2785: 3.0.4 Z Seam Alignment > "Random" option is with bug

Ando120, 2017-11-15:
Z Seam Alignment > "Random" option is with bug
Example with layer height 0.1 bug appears on layers 29-34
Dont travel along object("Combing Mode" is "All" )
Visible with some cylinder objects
[image]

smartavionics, 2017-11-15:
Looks like https://github.com/Ultimaker/CuraEngine/issues/625

smartavionics, 2017-11-15:
Hi @Ando120, if you set combing mode to "no skin" do the buggy travels go away?

Ando120, 2017-11-15:
With "no skin" travels are with retracton. So all is fine with "random"+ "no skin"
but I don't want to use it because too much retractions with some objects.

Made more tests and that height was related with that specific object.
Looks that bug appear only with thin wall.
[image]

If I redesign my object then I can avoid that bug.

Ghostkeeper, 2017-11-25:
I can't reproduce this bug. I'm using a little tube that I made to look like yours: [Tube.zip](https://github.com/Ultimaker/Cura/files/1503123/Tube.zip)
But it doesn't create any such travel moves through the object if Combing Mode is set to All.

Ando120, 2017-11-27:
Resize your object to 110% then it appear.

Ghostkeeper, 2017-12-05:
Indeed it does! Seems like we've found some good reproduce steps for https://github.com/Ultimaker/CuraEngine/issues/625 then. But I'll close this as a duplicate. Indeed it seems to be the same bug.

closed 2017-12-05 by Ghostkeeper