Ultimaker/Cura #2789: 3.0.3 and 3.0.4 stop communicating with the printer

HatTrick1914, 2017-11-15:
In both versions running on a Windows 10 machine and an Ultimaker 3 printer, both hardwire on a LAN, the "Monitor" function just stops reporting temperatures from the printer and there seems to be no way to recover from that other than restating Cura. 
Also the video feed will sporadically freeze. Most of the time if I switch from "Monitor" to "Prepare" and back again it will recover but on occasion I need to restart Cura.

diegopradogesto, 2017-11-15:
Could you please send us the log file just after noticing these issues? You can follow the instructions in the readme to get this file.

HatTrick1914, 2017-11-30:
I reset the log, started Cura and ran a print. I had a meeting and when I came back at 11:30 the monitor was frozen (not updating) The Cura log is attached.
[cura.log](https://github.com/Ultimaker/Cura/files/1518675/cura.log)

HatTrick1914, 2017-11-30:
I just clicked over to Prepare and back to monitor and now it is updating however it did not make any additional log entries from the one I previously attached. (attaching the latest after getting it to update again)
[cura.log](https://github.com/Ultimaker/Cura/files/1518689/cura.log)

Ghostkeeper, 2017-12-13:
Yeah the two log files seem to be identical. And no log entries are listed about the camera or connection other than the initial connection and authentication, so the log doesn't seem to help.

When the camera freezes that's usually a problem with the network connection. Even with a normal LAN connection it's normal that some packets get lost sometimes. However Cura should recover from this.

Your problem is kind of hard to reproduce... I've seen it last summer all the time but recently not so much ever since fieldOfView fixed the latest of that memory leak in the camera (and the firmware has improved as well). Maybe you could try updating Cura and the firmware?

There's not much that we can do if it's not happening on our end.

closed 2018-03-05 by ianpaschal