Ultimaker/Cura #2791: Noise in CuraEngineBackend

thopiekar, 2017-11-15:
I don't know why it comes, but I noticed while working on a plugin, that a lot of signals get emitted in our CuraEngine backend.
Even when there is no change on the value, I see no differences. So I wonder, why should a connected function to these signals care about a non-changed state?
If we need something that keeps the functions executed frequently, we can use timers.

The functions I connected these signals to, are looking basically like this:
```
def backendStateChange(self, state):
        Logger.log("d", "Sending backendStateChange event over COM with value: {}".format(state))
        self._BroadcastNotify(self.NotifySlicingDone, (state,))
```
    
Here is how I connected to these signals:
```
        self.backend = Application.getInstance().getBackend()
        self.backend.backendStateChange.connect(self.backendStateChange)
        self.backend.processingProgress.connect(self.backendProcessingProgress)        
```

```
2017-11-15 19:53:21,087 - DEBUG - CuraInterfacePlugin.InterfaceHandler.backendProcessingProgress [93]: Sending backendProcessingProgress event over COM with value: 0.0
2017-11-15 19:53:21,087 - DEBUG - CuraInterfacePlugin.InterfaceHandler.backendStateChange [86]: Sending backendStateChange event over COM with value: 1
2017-11-15 19:53:21,089 - DEBUG - CuraInterfacePlugin.InterfaceHandler.backendStateChange [86]: Sending backendStateChange event over COM with value: 1
2017-11-15 19:53:21,093 - DEBUG - CuraInterfacePlugin.InterfaceHandler.backendProcessingProgress [93]: Sending backendProcessingProgress event over COM with value: 0.0
2017-11-15 19:53:21,093 - DEBUG - CuraInterfacePlugin.InterfaceHandler.backendStateChange [86]: Sending backendStateChange event over COM with value: 1
2017-11-15 19:53:21,098 - DEBUG - CuraInterfacePlugin.InterfaceHandler.backendStateChange [86]: Sending backendStateChange event over COM with value: 1
2017-11-15 19:53:21,098 - DEBUG - CuraInterfacePlugin.InterfaceHandler.backendProcessingProgress [93]: Sending backendProcessingProgress event over COM with value: 0.0
2017-11-15 19:53:21,099 - DEBUG - CuraInterfacePlugin.InterfaceHandler.backendStateChange [86]: Sending backendStateChange event over COM with value: 1
2017-11-15 19:53:21,100 - DEBUG - CuraInterfacePlugin.InterfaceHandler.backendStateChange [86]: Sending backendStateChange event over COM with value: 1
2017-11-15 19:53:21,103 - DEBUG - CuraInterfacePlugin.InterfaceHandler.backendProcessingProgress [93]: Sending backendProcessingProgress event over COM with value: 0.0
2017-11-15 19:53:21,103 - DEBUG - CuraInterfacePlugin.InterfaceHandler.backendStateChange [86]: Sending backendStateChange event over COM with value: 1
2017-11-15 19:53:21,113 - DEBUG - CuraInterfacePlugin.InterfaceHandler.backendStateChange [86]: Sending backendStateChange event over COM with value: 1
2017-11-15 19:53:21,114 - DEBUG - CuraInterfacePlugin.InterfaceHandler.backendProcessingProgress [93]: Sending backendProcessingProgress event over COM with value: 0.0
2017-11-15 19:53:21,121 - DEBUG - CuraInterfacePlugin.InterfaceHandler.backendStateChange [86]: Sending backendStateChange event over COM with value: 1
2017-11-15 19:53:21,122 - DEBUG - CuraInterfacePlugin.InterfaceHandler.backendStateChange [86]: Sending backendStateChange event over COM with value: 1
2017-11-15 19:53:21,123 - DEBUG - CuraInterfacePlugin.InterfaceHandler.backendProcessingProgress [93]: Sending backendProcessingProgress event over COM with value: 0.0
2017-11-15 19:53:21,125 - DEBUG - CuraInterfacePlugin.InterfaceHandler.backendStateChange [86]: Sending backendStateChange event over COM with value: 1
2017-11-15 19:53:21,128 - DEBUG - CuraInterfacePlugin.InterfaceHandler.backendStateChange [86]: Sending backendStateChange event over COM with value: 1
2017-11-15 19:53:21,129 - DEBUG - CuraInterfacePlugin.InterfaceHandler.backendProcessingProgress [93]: Sending backendProcessingProgress event over COM with value: 0.0
2017-11-15 19:53:21,130 - DEBUG - CuraInterfacePlugin.InterfaceHandler.backendStateChange [86]: Sending backendStateChange event over COM with value: 1
2017-11-15 19:53:21,130 - DEBUG - CuraInterfacePlugin.InterfaceHandler.backendStateChange [86]: Sending backendStateChange event over COM with value: 1
2017-11-15 19:53:21,131 - DEBUG - CuraInterfacePlugin.InterfaceHandler.backendProcessingProgress [93]: Sending backendProcessingProgress event over COM with value: 0.0
2017-11-15 19:53:21,132 - DEBUG - CuraInterfacePlugin.InterfaceHandler.backendStateChange [86]: Sending backendStateChange event over COM with value: 1
2017-11-15 19:53:21,138 - DEBUG - CuraInterfacePlugin.InterfaceHandler.backendStateChange [86]: Sending backendStateChange event over COM with value: 1
2017-11-15 19:53:21,139 - DEBUG - CuraInterfacePlugin.InterfaceHandler.backendProcessingProgress [93]: Sending backendProcessingProgress event over COM with value: 0.0
2017-11-15 19:53:21,140 - DEBUG - CuraInterfacePlugin.InterfaceHandler.backendStateChange [86]: Sending backendStateChange event over COM with value: 1
2017-11-15 19:53:21,142 - DEBUG - CuraInterfacePlugin.InterfaceHandler.backendStateChange [86]: Sending backendStateChange event over COM with value: 1
2017-11-15 19:53:21,143 - DEBUG - CuraInterfacePlugin.InterfaceHandler.backendProcessingProgress [93]: Sending backendProcessingProgress event over COM with value: 0.0
2017-11-15 19:53:21,144 - DEBUG - CuraInterfacePlugin.InterfaceHandler.backendStateChange [86]: Sending backendStateChange event over COM with value: 1
2017-11-15 19:53:21,155 - DEBUG - CuraInterfacePlugin.InterfaceHandler.backendStateChange [86]: Sending backendStateChange event over COM with value: 1
2017-11-15 19:53:21,158 - DEBUG - CuraInterfacePlugin.InterfaceHandler.backendProcessingProgress [93]: Sending backendProcessingProgress event over COM with value: 0.0
2017-11-15 19:53:21,159 - DEBUG - CuraInterfacePlugin.InterfaceHandler.backendStateChange [86]: Sending backendStateChange event over COM with value: 1
2017-11-15 19:53:21,162 - DEBUG - CuraInterfacePlugin.InterfaceHandler.backendStateChange [86]: Sending backendStateChange event over COM with value: 1
2017-11-15 19:53:21,165 - DEBUG - CuraInterfacePlugin.InterfaceHandler.backendProcessingProgress [93]: Sending backendProcessingProgress event over COM with value: 0.0
2017-11-15 19:53:21,166 - DEBUG - CuraInterfacePlugin.InterfaceHandler.backendStateChange [86]: Sending backendStateChange event over COM with value: 1
2017-11-15 19:53:21,174 - DEBUG - CuraInterfacePlugin.InterfaceHandler.backendStateChange [86]: Sending backendStateChange event over COM with value: 1
2017-11-15 19:53:21,178 - DEBUG - CuraInterfacePlugin.InterfaceHandler.backendProcessingProgress [93]: Sending backendProcessingProgress event over COM with value: 0.0
2017-11-15 19:53:21,180 - DEBUG - CuraInterfacePlugin.InterfaceHandler.backendStateChange [86]: Sending backendStateChange event over COM with value: 1
2017-11-15 19:53:21,185 - DEBUG - CuraInterfacePlugin.InterfaceHandler.backendStateChange [86]: Sending backendStateChange event over COM with value: 1
2017-11-15 19:53:21,190 - DEBUG - CuraInterfacePlugin.InterfaceHandler.backendProcessingProgress [93]: Sending backendProcessingProgress event over COM with value: 0.0
2017-11-15 19:53:21,191 - DEBUG - CuraInterfacePlugin.InterfaceHandler.backendStateChange [86]: Sending backendStateChange event over COM with value: 1
2017-11-15 19:53:21,201 - DEBUG - CuraInterfacePlugin.InterfaceHandler.backendStateChange [86]: Sending backendStateChange event over COM with value: 1
2017-11-15 19:53:21,202 - DEBUG - CuraInterfacePlugin.InterfaceHandler.backendProcessingProgress [93]: Sending backendProcessingProgress event over COM with value: 0.0
2017-11-15 19:53:21,209 - DEBUG - CuraInterfacePlugin.InterfaceHandler.backendStateChange [86]: Sending backendStateChange event over COM with value: 1
2017-11-15 19:53:21,211 - DEBUG - CuraInterfacePlugin.InterfaceHandler.backendStateChange [86]: Sending backendStateChange event over COM with value: 1
2017-11-15 19:53:21,215 - DEBUG - CuraInterfacePlugin.InterfaceHandler.backendProcessingProgress [93]: Sending backendProcessingProgress event over COM with value: 0.0
2017-11-15 19:53:21,217 - DEBUG - CuraInterfacePlugin.InterfaceHandler.backendStateChange [86]: Sending backendStateChange event over COM with value: 1
2017-11-15 19:53:21,225 - DEBUG - CuraInterfacePlugin.InterfaceHandler.backendStateChange [86]: Sending backendStateChange event over COM with value: 1
[... cut ...]
2017-11-15 19:53:21,238 - DEBUG - CuraInterfacePlugin.InterfaceHandler.backendProcessingProgress [93]: Sending backendProcessingProgress event over COM with value: 0.0
2017-11-15 19:53:21,239 - DEBUG - CuraInterfacePlugin.InterfaceHandler.backendStateChange [86]: Sending backendStateChange event over COM with value: 1
2017-11-15 19:53:21,240 - DEBUG - CuraInterfacePlugin.InterfaceHandler.backendStateChange [86]: Sending backendStateChange event over COM with value: 1
```

Is it not in sense of events to emit them only if something really changed?

Ghostkeeper, 2017-11-25:
We do check if we've already sent the same progress message, down to 0.1% accuracy:

https://github.com/Ultimaker/CuraEngine/blob/f8697149514f0f7cc0677b7c2b426354321c7326/src/commandSocket.cpp#L611-L614

I just tested by putting a line below that if statement to log the progress:

    logAlways("Progress: %i", rounded_amount);

And I'm getting no duplicate amounts there.

thopiekar, 2017-11-25:
Strange. The log lines you see are there without a model loaded.

In general, the check whether the value has been already sent should already be done before emitting it.
I don't see a reason why the same value has to be sent all the time. Depending how many slots are connected to the signal, many functions might get called for no reason and most importantly no change.

At least this is what I expect from a event system, which works in sense of actio = reactio.
But to be honest, if I would write code with signals I would expect this check to be done by .emit(args*).
Maybe I will take a look on this again..

Ghostkeeper, 2017-12-01:
The code I pointed to is where it checks before emitting it.

thopiekar, 2017-12-01:
Excuse my laziness 😅 
So why is it still noisy in Cura? 🤔

Ghostkeeper, 2017-12-13:
Engines are often noisy :smile:

Maybe you're rounding somewhere? I added this code to [the function where it gets the progress message](https://github.com/Ultimaker/Cura/blob/7cd344978149997d91d91c1f20788e42fe3de91e/plugins/CuraEngineBackend/CuraEngineBackend.py#L484):

    print("Got progress message:", message.amount)

Then the output becomes this:

    <some very long engine command>
    Got progress message: 0.018530502915382385
    Got progress message: 0.023921897634863853
    Got progress message: 0.026447411626577377
    Got progress message: 0.02897292748093605
    Got progress message: 0.031498443335294724
    Got progress message: 0.0340239554643631
    Got progress message: 0.03654947131872177
    ...

Or maybe you're seeing places where the Python code sets the progress to 0 because it's starting a slice and resetting the engine and such...

closed 2018-03-02 by ChrisTerBeke