Ultimaker/Cura #2792: 3.04 Auto Layout raising model by first layer height

silycr, 2017-11-15:
Model is automatically placed 0.1mm above bed level.
Using supports and skirt
*First Layer
[image]
*Second Layer
[image]
*Z offset -0.1mm ; First Layer
[image]

Vandrasc, 2017-11-17:
Hi @silycr, I am not able to reproduce using a default profile. I noticed that you are using a customised profile, is it possible to link your project file so we can see what went wrong? Thank you!

closed 2017-11-25 by Ghostkeeper

Ghostkeeper, 2017-12-05:
Correction: Apparently that didn't make it into 3.1, so it'll be 3.2 then :(

BagelOrb, 2017-12-15:
Not all those are duplicates. Some are a similar problem caused by the engine.

The first layer might be too small to be printed. If another model does have a first layer then this model will be printed on the second layer.