Ultimaker/Cura #2793: If object is under the build plate no error message is flagged up (or object greyed out)

robinbon, 2017-11-16:
I have series of objects on build plate.

Inadvertently found 1 was below the build plate on printing.

Seems no error message is flagged up if this is the case.
[image]

nallath, 2017-11-16:
That's because it's not an error. This is used to print parts of the object. Everything below the build plate is ignored (and not sliced). This is quite usefull for scanned objects, which quite frequently don't have flat surfaces.

robinbon, 2017-11-16:
Ok, but thought that would be an option (bit like drop to build plate).

closed 2017-11-16 by ChrisTerBeke