Ultimaker/Cura #2794: Feature Request - Z-hop for ironing

Chompworks, 2017-11-16:
When ironing, there is still nozzle travel over previously ironed surfaces, slightly reducing the smooth result that ironing is meant to achieve.
Would it be possible to add a toggleable option for ironing travel moves to z-hop? [or is it tied to retractions z-hopping or a similar separate control elsewhere in Cura?]
[image]

smartavionics, 2017-11-16:
Hi, do you have combing enabled? If so, setting to "no skin" could stop that happening?

Chompworks, 2017-11-16:
Thats actually with no skin combing enabled - was present in Cura 2 and just remembered to test it in Cura 3 while dealing with a different issue.

smartavionics, 2017-11-16:
OK, can you please make the project available and then I can see where those travels are being created?

Chompworks, 2017-11-16:
Sure, here you go:
[MaterialSampleCura.3mf.zip](https://github.com/Ultimaker/Cura/files/1479220/MaterialSampleCura.3mf.zip)

smartavionics, 2017-11-16:
Thanks for the project. I notice two things:

1 - combing was set to all, you want it to be no-skin so that it retracts when moving over skin.

2 - you need to z-hop on retract otherwise it will just drag the nozzle.

Hope this helps.

Ghostkeeper, 2017-11-25:
The No-Skin combing setting is sort of meant for this.

GregValiant, 2024-11-23:
I'm cleaning house and checking to see if these old requests are still valid.
Can this be closed?

Chompworks, 2024-11-23:
> I'm cleaning house and checking to see if these old requests are still valid. Can this be closed?

I’ll be frank - I’ve stopped using Cura as a slicer for my printing workflows now, so it can be closed as far as I’m concerned.

closed 2024-11-23 by Chompworks

GregValiant, 2024-11-24:
I have a post-processor that accomplishes this by adding z-hops just for particular layers.  At some point I'll get it added to Cura.