Ultimaker/Cura #2795: 3.0.4: Priming at the end of the retraction happening too early

badvision, 2017-11-16:
I have a MakerSelect 3D, upgraded with a Ramps 1.4 running Repetier firmware and a microswiss hotend.  (there's other mods, but not important for this issue)

I have retraction and coasting enabled, and there's no stringing on the edges where the head starts retraction.  However, within 2mm or so of the next surface, I'm getting blobbing.  There's an extra priming amount setting that helps solve under extrusion at the end of retraction, but I see no way to control when priming starts, only the feed rate.  Is there any way to tell it to prime less material at the end of a retraction?

badvision, 2017-11-16:
[MS_Retraction Towers.curaproject.3mf.zip](https://github.com/Ultimaker/Cura/files/1479021/MS_Retraction.Towers.curaproject.3mf.zip)

Here is the model I am using to test my retraction settings.  Cura is printing the shapes in the same clockwise order, making it easier to see the priming issue.

badvision, 2017-11-16:
I think this is the same as #2667 -- A negative setting is yielding the same result as a zero-setting.

Ghostkeeper, 2017-11-25:
This is indeed a duplicate.

closed 2017-11-25 by Ghostkeeper