Ultimaker/Cura #2796: Plugin Settings not Saved

thesec, 2017-11-17:
Using post processing extensions I am all but getting used to have to set up these from scratch with every new project especially if it is just the model that I exchange or even just slightly tweaking a previously saved project.

Please save the extension/s' settings with the project file
or
at least save them when leaving the app.

fieldOfView, 2017-11-17:
This is by design. I never agreed with the design either.

The idea was that there were too many users that forgot they had postprocessing options enabled from the last session.

thesec, 2017-11-18:
... on the other hand all other is thrown in and even tried to mix machine settings from different projects ... all but consequent and all but good design with consistency being one of the base pillars of such.

One might think of integrating the post processing (and extension settings) with the extruder settings (being able to apply them per each ...) and as such making them transparent to the users and hence not risiking that they are forgotten about.

Getting this in as an option to advanced users likely to tweak more would also be an option.

Ghostkeeper, 2017-11-26:
We can't apply the post-processing settings for each extruder. They act on the final g-code, so they can even apply stuff cross-extruder, such as replace all `T1` commands by `M117 Do a little dance!`.

There's two main use cases for post-processing scripts: To fix some weirdness in the user's firmware that's incompatible with a preset g-code flavour (Search and Replace), and to make a tweak in the model (Tweak at Z or Pause at Height). For the first it would be handy to save it. For the second it wouldn't be handy at all. An option to save the post-processing script with the current printer can be added, but it'll have to be optional. I'll see what we can do about it.

thesec, 2017-11-26:
Thank you @Ghostkeeper .

Ghostkeeper, 2017-11-26:
See also https://github.com/nallath/PostProcessingPlugin/issues/46

thesec, 2017-11-26:
I see. Four+ totally separate settings' storages (Cura preferences, Printer settings, Extensions and the print settings) are just too much to handle and keep an eye on or even keep on mind ...

It would already be of great help to have all settings saved on exiting the application:
As a sample and reference I would like to point to Photoshop's extensions aka Filters where every single one's last settings are stored, despite affecting and being applicable on different part of the workflow, each.

ChrisTerBeke, 2018-07-30:
It seems this was fixed in #3229.

closed 2018-07-30 by ChrisTerBeke