Ultimaker/Cura #2798: master branch crash

smartavionics, 2017-11-17:
Just updated uranium, cura & curaengine master branches and now I get this:
````
2017-11-17 14:38:15,387 - DEBUG - UM.Controller.setActiveView [85]: Setting active view to SolidView
2017-11-17 14:38:15,691 - CRITICAL - cura.CrashHandler.__init__ [62]: An uncaught exception has occurred!
2017-11-17 14:38:15,694 - CRITICAL - cura.CrashHandler.__init__ [65]: Traceback (most recent call last):
2017-11-17 14:38:15,694 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/opt/cura/bin/cura", line 83, in <module>
2017-11-17 14:38:15,694 - CRITICAL - cura.CrashHandler.__init__ [65]:     app.run()
2017-11-17 14:38:15,695 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/opt/cura/lib/python3/dist-packages/cura/CuraApplication.py", line 683, in run
2017-11-17 14:38:15,695 - CRITICAL - cura.CrashHandler.__init__ [65]:     self._volume = BuildVolume.BuildVolume(root)
2017-11-17 14:38:15,695 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/opt/cura/lib/python3/dist-packages/cura/BuildVolume.py", line 107, in __init__
2017-11-17 14:38:15,695 - CRITICAL - cura.CrashHandler.__init__ [65]:     Application.getInstance().getMachineManager().activeQualityChanged.connect(self._onStackChanged)
2017-11-17 14:38:15,695 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/opt/cura/lib/python3/dist-packages/cura/CuraApplication.py", line 735, in getMachineManager
2017-11-17 14:38:15,695 - CRITICAL - cura.CrashHandler.__init__ [65]:     self._machine_manager = MachineManager.createMachineManager()
2017-11-17 14:38:15,696 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/opt/cura/lib/python3/dist-packages/cura/Settings/MachineManager.py", line 1215, in createMachineManager
2017-11-17 14:38:15,696 - CRITICAL - cura.CrashHandler.__init__ [65]:     return MachineManager()
2017-11-17 14:38:15,696 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/opt/cura/lib/python3/dist-packages/cura/Settings/MachineManager.py", line 109, in __init__
2017-11-17 14:38:15,696 - CRITICAL - cura.CrashHandler.__init__ [65]:     self.setActiveMachine(active_machine_id)
2017-11-17 14:38:15,696 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/opt/cura/lib/python3/dist-packages/cura/Settings/MachineManager.py", line 357, in setActiveMachine
2017-11-17 14:38:15,697 - CRITICAL - cura.CrashHandler.__init__ [65]:     Application.getInstance().setGlobalContainerStack(containers[0])
2017-11-17 14:38:15,697 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/opt/cura/lib/python3/dist-packages/UM/Application.py", line 154, in setGlobalContainerStack
2017-11-17 14:38:15,697 - CRITICAL - cura.CrashHandler.__init__ [65]:     self.globalContainerStackChanged.emit()
2017-11-17 14:38:15,697 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/opt/cura/lib/python3/dist-packages/UM/Signal.py", line 212, in emit
2017-11-17 14:38:15,697 - CRITICAL - cura.CrashHandler.__init__ [65]:     self.__performEmit(*args, **kwargs)
2017-11-17 14:38:15,697 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/opt/cura/lib/python3/dist-packages/UM/Signal.py", line 315, in __performEmit
2017-11-17 14:38:15,698 - CRITICAL - cura.CrashHandler.__init__ [65]:     func(dest, *args, **kwargs)
2017-11-17 14:38:15,698 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/opt/cura/lib/python3/dist-packages/cura/BuildVolume.py", line 500, in _onStackChanged
2017-11-17 14:38:15,698 - CRITICAL - cura.CrashHandler.__init__ [65]:     self._updateDisallowedAreas()
2017-11-17 14:38:15,698 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/opt/cura/lib/python3/dist-packages/cura/BuildVolume.py", line 597, in _updateDisallowedAreas
2017-11-17 14:38:15,698 - CRITICAL - cura.CrashHandler.__init__ [65]:     disallowed_border_size = self._getEdgeDisallowedSize()
2017-11-17 14:38:15,698 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/opt/cura/lib/python3/dist-packages/cura/BuildVolume.py", line 956, in _getEdgeDisallowedSize
2017-11-17 14:38:15,699 - CRITICAL - cura.CrashHandler.__init__ [65]:     move_from_wall_radius = max(move_from_wall_radius, max(self._getSettingFromAllExtruders("infill_wipe_dist")))
2017-11-17 14:38:15,699 - CRITICAL - cura.CrashHandler.__init__ [65]: ValueError: max() arg is an empty sequence
````

smartavionics, 2017-11-17:
The crash is caused by something from fb87b73 to 01437b2 my guess is something in f48539c is causing the problem?

LipuFei, 2017-11-20:
@smartavionics Is your currently active machine a single-extrusion machine? In this PR #2685, all machines will have >= 1 ExtruderStacks, even for single-extrusion machines. From the log, looks like single extrusion machines are not upgraded properly. (probably because you were running master?)

smartavionics, 2017-11-20:
Hi, yes, it's a single extruder printer. Can I upgrade manually?

LipuFei, 2017-11-20:
I'm not sure if the file upgrade works yet... I will get back to you later.

smartavionics, 2017-11-20:
OK, no problem.

LipuFei, 2017-11-20:
@smartavionics Could you try this?
1. Open your cura folder (should be ".local/share/cura") and find the machine file in the "machine_instances" folder.
2. Open the machine file and set "setting_version" to 3.
3. Start Cura again.

smartavionics, 2017-11-20:
Great, that worked, I changed the setting_version on two of my printers (Kossel XL and Kossel Mini) and cura started up OK. Here's what came out on the terminal if you are interested:
````
QApplication: invalid style override passed, ignoring it.
UM.Resources.__initializeStoragePaths [342]: Initializing storage paths
UM.Resources.__initializeStoragePaths [353]: Config storage path is /home/markb/.config/cura/master
UM.Resources.__initializeStoragePaths [361]: Data storage path is /home/markb/.local/share/cura/master
UM.Resources.__initializeStoragePaths [373]: Cache storage path is /home/markb/.cache/cura/master
UM.View.GL.OpenGLContext.detectBestOpenGLVersion [103]: Trying OpenGL context 4.1...
UM.View.GL.OpenGLContext.detectBestOpenGLVersion [113]: Yay, we got at least OpenGL 4.1 core: 4.5 Core profile
UM.Qt.QtApplication.__init__ [82]: Detected most suitable OpenGL context version: 4.1 Core profile
UM.Qt.QtApplication.__init__ [98]: Command line arguments: {'qmljsdebugger': None, 'external-backend': False, 'single_instance': False, 'disable-textures': False, 'headless': False, 'file': []}
2017-11-20 08:06:43,284 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin ConsoleLogger
2017-11-20 08:06:43,313 - INFO - CuraEngineBackend.CuraEngineBackend.__init__ [64]: Found CuraEngine at: /opt/cura/bin/CuraEngine
2017-11-20 08:06:43,313 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin CuraEngineBackend
2017-11-20 08:06:43,319 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin ScaleTool
2017-11-20 08:06:43,322 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin TranslateTool
2017-11-20 08:06:43,327 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin MirrorTool
2017-11-20 08:06:43,329 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin SelectionTool
2017-11-20 08:06:43,332 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin CameraTool
2017-11-20 08:06:43,336 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin RotateTool
2017-11-20 08:06:43,343 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin STLReader
2017-11-20 08:06:43,347 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin OBJReader
2017-11-20 08:06:43,349 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin STLWriter
2017-11-20 08:06:43,352 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin OBJWriter
2017-11-20 08:06:43,356 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin LocalFileOutputDevice
2017-11-20 08:06:43,360 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin SimpleView
2017-11-20 08:06:43,360 - WARNING - UM.PluginRegistry.loadPlugin [200]: Plugin ConsoleLogger was already loaded
2017-11-20 08:06:43,362 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin FileLogger
2017-11-20 08:06:43,418 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin UpdateChecker
2017-11-20 08:06:43,430 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin SliceInfoPlugin
2017-11-20 08:06:43,439 - INFO - UpdateChecker.UpdateCheckerJob.run [34]: Checking for new version of cura
2017-11-20 08:06:43,444 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin SolidView
2017-11-20 08:06:43,451 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin ImageReader
2017-11-20 08:06:43,458 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin PerObjectSettingsTool
2017-11-20 08:06:43,466 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin RemovableDriveOutputDevice
2017-11-20 08:06:43,470 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin UserAgreementPlugin
2017-11-20 08:06:43,475 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin GCodeProfileReader
2017-11-20 08:06:43,484 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin PluginBrowser
2017-11-20 08:06:43,488 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin VersionUpgrade30to31
2017-11-20 08:06:43,493 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin VersionUpgrade22to24
2017-11-20 08:06:43,497 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin VersionUpgrade21to22
2017-11-20 08:06:43,499 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin VersionUpgrade27to30
2017-11-20 08:06:43,502 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin VersionUpgrade26to27
2017-11-20 08:06:43,506 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin VersionUpgrade25to26
2017-11-20 08:06:43,520 - DEBUG - UM.PluginRegistry.loadPlugin [206]: Plugin USBPrinting was disabled
2017-11-20 08:06:43,520 - WARNING - UM.PluginRegistry.loadPlugin [200]: Plugin CuraEngineBackend was already loaded
2017-11-20 08:06:43,530 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin X3DReader
2017-11-20 08:06:43,534 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin AutoSave
2017-11-20 08:06:43,540 - DEBUG - UM.PluginRegistry.loadPlugin [206]: Plugin FirmwareUpdateChecker was disabled
2017-11-20 08:06:43,546 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin XRayView
2017-11-20 08:06:43,559 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin LayerView
2017-11-20 08:06:43,573 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin XmlMaterialProfile
2017-11-20 08:06:43,648 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin UM3NetworkPrinting
2017-11-20 08:06:43,652 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin CuraProfileWriter
[... cut ...]
2017-11-20 08:08:13,820 - DEBUG - UM.Mesh.MeshData.approximateConvexHull [387]: approximateConvexHull(target_count=1024) Calculating 3D convex hull took 0.012406349182128906 seconds. 79 input vertices. 79 output vertices.
file:///opt/cura/lib/qml/QtQuick/Controls/TextField.qml:638:5: QML TextInputWithHandles: Binding loop detected for property "text"
2017-11-20 08:08:13,907 - DEBUG - UM.View.GL.ShaderProgram.load [55]: Loading shader file [/opt/cura/share/cura/resources/shaders/transparent_object.shader]...
2017-11-20 08:08:14,172 - DEBUG - UM.Mesh.MeshData.calculateNormalsFromIndexedVertices [443]: Calculating normals took 0.0024895668029785156 seconds
````

LipuFei, 2017-11-20:
There is still some problems in the version upgrade... Those `version outdated` warning messages shouldn't occur. If you don't mind, could you pack your cura directory and send it to me? Thanks in advance!

LipuFei, 2017-11-20:
@smartavionics Ah, now I know... Could you also downgrade all the extruder files too? just like the machine files. They are in the "extruders" folder.

smartavionics, 2017-11-20:
Here's my master config folder. It's probably a real mess because of all the upgrades/changes that have been made as the code evolves!

[mb-cura-master.zip](https://github.com/Ultimaker/Cura/files/1486923/mb-cura-master.zip)

smartavionics, 2017-11-20:
I did downgrade the extruders as you requested. The archive contains the files after I had downgraded and then run cura.

LipuFei, 2017-11-20:
@smartavionics Do you still see any of those "outdated" warning messages? They should be gone now.

smartavionics, 2017-11-20:
Yes, the're still there. But they are not a recent thing, I have been seeing those messages for quite a while now (at least weeks).

LipuFei, 2017-11-20:
Those were built-in "Not Supported" quality profiles, which are now removed. I will test with your config folder. Thanks.

smartavionics, 2017-11-20:
Hmm, I just realised that what I did this morning (resetting setting_version to 3 and using your latest code) made all my existing profiles go away and also all the materials went away. So I have reverted back to what  I know works.

LipuFei, 2017-11-20:
Could you pack your currently working cura folder and send it to us? The previous package wasn't very useful...

smartavionics, 2017-11-20:
Sure, do you mean the .local/cura folder or anything else?

LipuFei, 2017-11-20:
That's the one :)

smartavionics, 2017-11-20:
OK, here's an updated zip, it even has a log file in it that shows some errors. 
[mb-cura-master.zip](https://github.com/Ultimaker/Cura/files/1487643/mb-cura-master.zip)

smartavionics, 2017-11-21:
Hi, any thoughts on this? I am still rather stuck because I can't use an master branch rev that is after your branch was merged (f48539c). The problems I am seeing are that all of my existing profiles go away and if I try and select another printer, cura crashes with a whole bunch of empty dialogues appearing on the screen. So, I'm screwed, really.

LipuFei, 2017-11-21:
Hi Mark, I have tried downgrading all the files in your `cura` folder and things seemed to be working. No file was lost after the upgrade. Could you try the same?

I have a script here which can downgrade all the `setting_version`s to 3:
```
#!/bin/sh

new_setting_version="3"
grep -r -n setting_version ./ | cut -d':' -f1 | xargs sed -i 's/setting_version = .*/setting_version = '"${new_setting_version}"'/g'
```

`cd` to your `cura` folder and run this script.

smartavionics, 2017-11-21:
It's no good, I run your script (thanks for that) and I can see that the setting_version value has changed in the files but I am still getting this crash:

````
2017-11-21 13:04:40,300 - DEBUG - UM.Controller.setActiveView [85]: Setting active view to SolidView
2017-11-21 13:04:41,226 - CRITICAL - cura.CrashHandler.__init__ [62]: An uncaught exception has occurred!
2017-11-21 13:04:41,228 - CRITICAL - cura.CrashHandler.__init__ [65]: Traceback (most recent call last):
2017-11-21 13:04:41,229 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/opt/cura/bin/cura", line 83, in <module>
2017-11-21 13:04:41,229 - CRITICAL - cura.CrashHandler.__init__ [65]:     app.run()
2017-11-21 13:04:41,229 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/opt/cura/lib/python3/dist-packages/cura/CuraApplication.py", line 683, in run
2017-11-21 13:04:41,229 - CRITICAL - cura.CrashHandler.__init__ [65]:     self._volume = BuildVolume.BuildVolume(root)
2017-11-21 13:04:41,229 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/opt/cura/lib/python3/dist-packages/cura/BuildVolume.py", line 107, in __init__
2017-11-21 13:04:41,230 - CRITICAL - cura.CrashHandler.__init__ [65]:     Application.getInstance().getMachineManager().activeQualityChanged.connect(self._onStackChanged)
2017-11-21 13:04:41,230 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/opt/cura/lib/python3/dist-packages/cura/CuraApplication.py", line 735, in getMachineManager
2017-11-21 13:04:41,230 - CRITICAL - cura.CrashHandler.__init__ [65]:     self._machine_manager = MachineManager.createMachineManager()
2017-11-21 13:04:41,230 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/opt/cura/lib/python3/dist-packages/cura/Settings/MachineManager.py", line 1215, in createMachineManager
2017-11-21 13:04:41,230 - CRITICAL - cura.CrashHandler.__init__ [65]:     return MachineManager()
2017-11-21 13:04:41,230 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/opt/cura/lib/python3/dist-packages/cura/Settings/MachineManager.py", line 109, in __init__
2017-11-21 13:04:41,231 - CRITICAL - cura.CrashHandler.__init__ [65]:     self.setActiveMachine(active_machine_id)
2017-11-21 13:04:41,231 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/opt/cura/lib/python3/dist-packages/cura/Settings/MachineManager.py", line 357, in setActiveMachine
2017-11-21 13:04:41,231 - CRITICAL - cura.CrashHandler.__init__ [65]:     Application.getInstance().setGlobalContainerStack(containers[0])
2017-11-21 13:04:41,231 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/opt/cura/lib/python3/dist-packages/UM/Application.py", line 154, in setGlobalContainerStack
2017-11-21 13:04:41,231 - CRITICAL - cura.CrashHandler.__init__ [65]:     self.globalContainerStackChanged.emit()
2017-11-21 13:04:41,232 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/opt/cura/lib/python3/dist-packages/UM/Signal.py", line 212, in emit
2017-11-21 13:04:41,232 - CRITICAL - cura.CrashHandler.__init__ [65]:     self.__performEmit(*args, **kwargs)
2017-11-21 13:04:41,232 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/opt/cura/lib/python3/dist-packages/UM/Signal.py", line 315, in __performEmit
2017-11-21 13:04:41,232 - CRITICAL - cura.CrashHandler.__init__ [65]:     func(dest, *args, **kwargs)
2017-11-21 13:04:41,232 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/opt/cura/lib/python3/dist-packages/cura/BuildVolume.py", line 500, in _onStackChanged
2017-11-21 13:04:41,232 - CRITICAL - cura.CrashHandler.__init__ [65]:     self._updateDisallowedAreas()
2017-11-21 13:04:41,233 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/opt/cura/lib/python3/dist-packages/cura/BuildVolume.py", line 597, in _updateDisallowedAreas
2017-11-21 13:04:41,233 - CRITICAL - cura.CrashHandler.__init__ [65]:     disallowed_border_size = self._getEdgeDisallowedSize()
2017-11-21 13:04:41,233 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/opt/cura/lib/python3/dist-packages/cura/BuildVolume.py", line 956, in _getEdgeDisallowedSize
2017-11-21 13:04:41,233 - CRITICAL - cura.CrashHandler.__init__ [65]:     move_from_wall_radius = max(move_from_wall_radius, max(self._getSettingFromAllExtruders("infill_wipe_dist")))
2017-11-21 13:04:41,233 - CRITICAL - cura.CrashHandler.__init__ [65]: ValueError: max() arg is an empty sequence
````

Here's the log file:

[cura.log.zip](https://github.com/Ultimaker/Cura/files/1491400/cura.log.zip)

LipuFei, 2017-11-21:
I noticed that you have some custom machine definitions. Could you send those to me too? Some changes might be needed.

smartavionics, 2017-11-21:
Here they are...

[mb-printer-defs.zip](https://github.com/Ultimaker/Cura/files/1491554/mb-printer-defs.zip)

LipuFei, 2017-11-21:
Hi Mark, I have fixed the problem with your Kossel XL. You can use this cura folder package: 
[mb-master-fix_01.zip](https://github.com/Ultimaker/Cura/files/1491691/mb-master-fix_01.zip)

Changes:
- Changed all `setting_version`s to `3`
- Fixed the  definition container in the Kossel XL stack

If you successfully start Cura, you will still have the problem of not seeing any custom quality profiles. We are working on this at the moment and will let you know when it's done.

LipuFei, 2017-11-21:
One thing to add: your custom profiles are there, they are just not shown on the GUI.

LipuFei, 2017-11-21:
Hi @smartavionics , could you pull the latest `master` of Cura and Uranium and test it out? It should work now.

smartavionics, 2017-11-21:
Thanks, I think it is fixed now. I can start cura without it crashing and my profiles have reappeared and I can switch between printers OK.

LipuFei, 2017-11-21:
@smartavionics FYI, there is still some issues with custom profiles... WIP

smartavionics, 2017-11-21:
OK

ChrisTerBeke, 2017-11-21:
Custom profiles were just fixed by https://github.com/Ultimaker/Cura/commit/0625d22e04e00e109ace8855074a3ab687746895. Would be great if you can try it out!

smartavionics, 2017-11-21:
I tried that and as far as I can tell, it worked OK. I can see the machines and profiles and switch between them.

ChrisTerBeke, 2017-12-10:
Re-opening as more people seem to have the error `ValueError: max() arg is an empty sequence`.

Thisismydigitalself, 2017-12-11:
After having many issues with 3.1 I am now back on 3.0.4, but it wont open my saved projects which were probably created on  3.1. Not only that but suddenly my machine settings got reset and lost and now i don't have my printer settings anymore.

i don't know what to do. this is becoming a serious issue.

fieldOfView, 2017-12-11:
Opening a 3.1 project in 3.0 sounds like a rather bad idea, with the potential to screw up your 3.0 configuration. Cura should have a safeguard against doing that.

Thisismydigitalself, 2017-12-11:
Trust me to not wanting to be in this situation. 
I am loosing time and money over this. my mistake for trying out 3.1 beta. 
learned my lesson.

ChrisTerBeke, 2017-12-11:
It won't indeed open a newer project file (we check versions). Beta's are absolutely meant as Beta. We always recommend using the Beta alongside the previous stable (those configs can live next to each other, no problem) and not using it for any serious work. That being said this Beta had a bit more issues that we'd like ourselves...

Thisismydigitalself, 2017-12-11:
Just reinstalled 3.1 beta and unlike the released version of 3.1, the beta runs and opens both my printer settings and projects. I hope this information will help in releasing a hotfix or 3.2 soon.

filipgoc, 2017-12-17:
> Cura should have a safeguard against doing that.
Right!

drewoprea, 2017-12-20:
I'm getting crashes just like you ( smartavionics ) describe: "cura crashes with a whole bunch of empty dialogues appearing on the screen." It happens some of the time when cura tries slicing a model.
I'm running Ver 3.1.0 on linux
last thing that gets written to the console is:
file:///tmp/.mount_NqVsw3/usr/bin/resources/themes/cura-light/styles.qml:197:17: QML Item: Binding loop detected for property "width"
or:
file:///tmp/.mount_wzEPtT/usr/bin/qt/qml/QtQuick/Controls/TextField.qml:638:5: QML TextInputWithHandles: Binding loop detected for property "text"

Let me know if this looks like the same bug/behavior or if i should file separately.

smartavionics, 2017-12-20:
Hello @drewoprea , I don't think those binding loop messages are necessarily bad because you get them anyway even when cura starts OK. Can you please find your cura.log file (probably in ~/.local/share/cura/3.1 or similar) and post that (or a link to a copy), then the devs can maybe see what's happening.

ChrisTerBeke, 2017-12-20:
Binding loops are just QT detecting that a property is being set from multiple sources or many times, and then blocking that to prevent more issues. They're not nice (as they're warnings), but not critical either. The current ones we get we couldn't figure out yet (where the property set loop is coming from).

drewoprea, 2017-12-20:
Here's cura.log
I've found that restarting my laptop fixes the crashes for a while.
Does this log get overwritten every time I run Cura? or is this log likely to contain the conditions surrounding my dozen or so crashes?

Thanks for the quick interest!
-Drew
[cura.log](https://github.com/Ultimaker/Cura/files/1575947/cura.log)

smartavionics, 2017-12-20:
Hi, I took a quick look at the file, it's cumulative so it contains all your old crashes! I think the messages of interest look like this:
````
2017-12-19 22:35:41,312 - CRITICAL - cura.CrashHandler.__init__ [62]: An uncaught exception has occurred!
2017-12-19 22:35:41,316 - CRITICAL - cura.CrashHandler.__init__ [65]: Traceback (most recent call last):
2017-12-19 22:35:41,319 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/build/3.1/build/inst/lib/python3.5/site-packages/cura/CuraApplication.py", line 770, in eve
nt
2017-12-19 22:35:41,322 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/build/3.1/build/inst/lib/python3.5/site-packages/UM/Qt/QtApplication.py", line 297, in even
t
2017-12-19 22:35:41,325 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/build/3.1/build/inst/lib/python3.5/site-packages/UM/Event.py", line 189, in call
2017-12-19 22:35:41,328 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/build/3.1/build/inst/lib/python3.5/site-packages/UM/Signal.py", line 315, in __performEmit
2017-12-19 22:35:41,330 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/build/3.1/build/inst/lib/python3.5/site-packages/UM/Backend/Backend.py", line 166, in _onSo
cketStateChanged
2017-12-19 22:35:41,333 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/build/3.1/build/inst/lib/python3.5/site-packages/UM/Backend/Backend.py", line 72, in startE
ngine
2017-12-19 22:35:41,336 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/build/3.1/build/inst/lib/python3.5/site-packages/UM/Backend/Backend.py", line 142, in _runE
ngineProcess
2017-12-19 22:35:41,338 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/build/env/3.1/inst/lib/python3.5/subprocess.py", line 947, in __init__
2017-12-19 22:35:41,341 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/build/env/3.1/inst/lib/python3.5/subprocess.py", line 1490, in _execute_child
2017-12-19 22:35:41,343 - CRITICAL - cura.CrashHandler.__init__ [65]: OSError: [Errno 12] Cannot allocate memory
````

Looks like it is running out of memory?
````

drewoprea, 2017-12-20:
That's possible, explains why rebooting fixes it, and suggests the few times I've had the laptop hang while working were also related.  This condition persisted after i restarted Cura.  Is it possible that there's memory not being freed on exit? or could this be off in python somewhere, i should go double check version and such?

smartavionics, 2017-12-20:
When the process exits all the normal memory should get freed but what may be happening is that there is some memory associated with the graphics that is getting used up and it does not get freed when the process exits. I know that on a previous Linux machine (coincidentally, also a laptop) that after running for some time (weeks) various programs would exhibit graphical glitches and rebooting always sorted it out. In fact, possibly just restarting the X server was sufficient, not sure about that.

smartavionics, 2018-01-15:
I think the original issue has been solved so closing...

closed 2018-01-15 by smartavionics