Ultimaker/Cura #2799: UM2 Crashes into metal build plate bracket

martinjacala, 2017-11-17:
I sliced my stl with latest cura 3.0.4 with pretty much basic settings. When i tried to print it i realised my UM2 crashed into the left glass plate bracked mid-print of the first layer. 

I have then tried to check the gcode and indeed, there is a strange move towards left the corner.
Attached is the gcode and the cura profile i used. I can also attach the stl if needed.

[image]
[Archive.zip](https://github.com/Ultimaker/Cura/files/1482762/Archive.zip)

smartavionics, 2017-11-17:
Hi, this could possibly be related to #2704 which has been fixed (I hope). As a workaround, if you slightly reduce the line width, the erroneous moves may go away. This worked for the other model that was showing a similar problem. I hope this helps.

Ghostkeeper, 2017-11-26:
Could you try this again with the 3.1 beta (or wait for the 3.1 release in 2 weeks)?

martinjacala, 2017-11-27:
I tried to slice the same file with same profile in 3.1 beta and the issue **is** gone! Thank you!

closed 2017-11-27 by ChrisTerBeke