Ultimaker/Cura #2803: Rafts not drawing in on bed with multiple objects

sshrum01, 2017-11-18:
Cura 3.0.4 - Windows 10

When loading multiple object on the print area with rafting on, the slicer is not rafting ALL items.

As a result, when the print is started, the slicer "acts" like a raft layer is put down and prints above a blank area for those objects that are missing the physical raft.  This issue is visible on screen in layer view.

[image]

diegopradogesto, 2017-11-20:
Have you enable some "Per object" settings? Are all the models laying down the build plate? Could you share your project file with us?

Ghostkeeper, 2017-11-28:
Build Plate Adhesion Type is a global setting (or rather, a resolved one), so it's impossible to set it per-object. I think some of sshrum01's objects are probably not flat on the build plate, but we won't know until he shares a project file or at least where he got the models from.
The bug doesn't occur for me. There's a proper raft around every object.

diegopradogesto, 2017-11-28:
I know the raft is a global setting, just asking if "some" per-object settings that can affect were enabled.
Let's wait for the project file and see what this issue is.

benjaminwand, 2017-11-29:
I have the same problem, project files attached. Or, what else do you need?

[drucken neu.zip](https://github.com/Ultimaker/Cura/files/1513828/drucken.neu.zip)

diegopradogesto, 2017-11-29:
If you can please send the Cura Project, is more useful because we can simulate exactly what you have (printer, material, settings, ...). Also please send the Cura version that you are using.
Thanks.

benjaminwand, 2017-11-29:
2.7.0
[font test druck.3mf.zip](https://github.com/Ultimaker/Cura/files/1514167/font.test.druck.3mf.zip)
It did not happen with Cura 3.

diegopradogesto, 2017-12-01:
Sorry, this is not the project, it's just a 3MF containing the models. The project should be *.curaproject.3mf and contains all the settings that you are using with those models. If you have doubts, please take a look to the user manual: https://ultimaker.com/en/resources/38954-cura-project-saving

benjaminwand, 2017-12-02:
Couldn't reproduce it again, sorry!

Ghostkeeper, 2017-12-15:
Benjaminwand couldn't reproduce it again, but does @sshrum01 have more information for us?

benjaminwand, 2018-03-14:
I had it again today, with Cura 3.1.0. I hope this file helps?
[UM3_0.03.3.3mf.zip](https://github.com/Ultimaker/Cura/files/1811368/UM3_0.03.3.3mf.zip)