Ultimaker/Cura #2804: 3.04 Unable to force Z hop

siwyroot, 2017-11-18:
Hello,

I'm trying to configure Z hope during traveling to next point but cura ignores all options regarding hope. Following options are set:

```
Enable retraction:				on
Retract at layer change: 	 		on
Retraction distance:				3
Retraction speed:				200m/s
Retraction minimum travel:			2mm
Maximum retraction count:			90
Retract before outer wall:			on
Avoid printed parts when traveling:		off
Z hop when retracted:				on
Z hop height:					1mm
```
I have try to change "Retraction minimum travel" to 0 and reinstalled cura but no luck. No matter what I change I cannot make it hop :/

BR

smartavionics, 2017-11-18:
If combing is enabled, it will not retract when travel is only within a single part. It should still retract when moving between parts.

siwyroot, 2017-11-18:
Hello,

Thanks for quick comment, option "Combing Mode" is set to All after changing it to off I can see "hop" in gcode. Retraction is happening but hop is not happening at all with "Combing Mode" set to all.

Thanks again, I was almost loosing it :)

smartavionics, 2017-11-18:
> Retraction is happening but hop is not happening at all with "Combing Mode" set to all.

It should not retract when moving within a part if combing is set to all. That's the whole point of combing. I normally use combing set to "no skin" as I don't like the non-retract travel moves leaving lines on the skin surfaces. For PETG I don't use combing at all as I find that the nozzle gets rather grubby and it leaves burnt blobs of plastic here and there.

siwyroot, 2017-11-18:
Well it's retracting at layer change with "Combing Mode" set to all. I was watching extruder during that time to confirm.

smartavionics, 2017-11-18:
> Well it's retracting at layer change with "Combing Mode" set to all. I was watching extruder during that time to confirm.

Yes, it will still retract at layer change (if enabled) and also when moving over non-combed areas. But within a combed area it will not retract (or z-hop).

siwyroot, 2017-11-18:
Thanks for detailed information. 

Best regards

closed 2017-11-18 by siwyroot

smartavionics, 2017-11-18:
You're welcome, happy printing!