Ultimaker/Cura #2805: Internal holes are narrower than the external dimensions of the model

voice2001, 2017-11-18:
Internal holes are narrower than the external dimensions of the model.
For example, I make a cube with a round hole in the middle. The dimensions of the cube are 20x20x20mm, the hole diameter is 8mm.
When printing, the cube is exactly 20x20x20mm. But a hole with a diameter of 7.5-7.8mm
This happens on the nozzle 0.4mm, and on the nozzle 1.0mm
What am I doing wrong?

ChrisTerBeke, 2017-11-19:
Due to layer width we always have to round off some lines that can cause these faulty tolerances. When designing for 3D printing that has to be taken into account. The new 3.1 version will have a slicing tolerance feature however that allows you to change this behaviour a bit, but it only works in (angled) Z direction for now.

Thisismydigitalself, 2017-11-19:
Any ETA for 3.1?
Can i download beta and try out now?

ChrisTerBeke, 2017-11-19:
Beta is scheduled for coming week.

maukcc, 2017-11-20:
@voice2001 is it the same 7.6mm with both the 0.4 and 1.0 nozzle ?
or is it 7.6 with the 0.4 and 7.0 with the 1.0 ?

voice2001, 2017-11-20:
I need to change the nozzle, which I could say for sure. Unfortunately I can not do this right now, but I'll try to check as soon as possible

Ghostkeeper, 2017-11-27:
The beta is out now, so you can try, but I don't think it'll help.

The actual reason why this is happening is not really the slicing tolerance. The slicing tolerance just determines whether the sliced layers should stay within the model's volume or try to approximate the volume as closely as possible (accepting some material outside of the volume to average it out). I think the name of that setting is a bit of a misnomer since it's not designed to allow the user to specify any tolerances.

The reason why this is happening and why this effect exists specifically on holes and not on perimeters is because the line gets dragged along with the movement of the nozzle. As the nozzle moves into a corner, the last millimetre or so of the previous line segment gets dragged along, still being fluid. This causes the wire to take a short cut in the corner.

This is the reason why the "Post stretch script" post-processing script was added, actually. It compensates for this effect in inside corners. I suggest you try that, but be careful because it tends to create some weirdness if you set it to extremes.

ianpaschal, 2018-03-09:
Question is answered and as @ChrisTerBeke said, these sorts of limitations must simply be taken into account when creating the original model. That's not 3D printing specific actually: all drafting (including the computer-aided variety) must account for how materials behave during the manufacturing process.

closed 2018-03-09 by ianpaschal