Ultimaker/Cura #2807: Printer's Start & End gcode

Thisismydigitalself, 2017-11-19:
today and out of the blue my printer's start/end gcode got changed. don't ask me how. Cura 3.0.4 just woke up this way. i then switched to 2.7 and i see the same. somehow it got changed and i have no clue how and why. not only that but also when i opened 3MF files and sent to my printer i saw the name of the file was not the same as my 3mf file. very strange. closing all cura instances and reopened and now the file name is correct but the start/end gcode is not. 

What cura wants from me?
what could have caused my printer's start/end code to change and where can i find my lost settings?

Thisismydigitalself, 2017-11-19:
For what it worth, i saw in Printers list a new printer " "My Printer Name" #2 " alongside my "my Printet name" printer. somehow in both 2.7 and 3.0.4 there was a new printer which i never created - same name with #2 sufix.

Can you tell what went wrong?

fieldOfView, 2017-11-19:
This happens when you load a project. There’s a dialog that asks if you want to update your current printer or add a new printer with the settings from the project. I don’t know how that would reset your start gcode though.

Thisismydigitalself, 2017-11-19:
I see what you mean. i use projects all the time. i **never** change the pull down menus - i just open those. i have no idea what happened this time and why my start/end gcode changed.  I have multiple users on this PC. I don't know if that was the case, but can an instance of Cura running on a different user caused this?

Ghostkeeper, 2017-11-27:
These settings are stored in the roaming user profile. They should be personal to your user profile.

To me it's a bit weird that it was also changed in 2.7. That means it must either have happened before you ever launched 3.0, or it happened again in 2.7 in exactly the same way. We never transfer settings from a newer version to an older version.

closed 2018-03-02 by ChrisTerBeke

Thisismydigitalself, 2018-03-23:
Cura 3.2.1.
Last night and while printing I noticed my printer did not start to print as usual.
checking the start code and again Cura, out of the blue,  changed it.
This is an on-going issue which looks like a bug but i don't even know where and how to reproduce it.

Thisismydigitalself, 2018-03-23:
Cura 3.2.1.
Last night and while printing I noticed my printer did not start to print as usual.
checking the start code and again Cura, out of the blue,  changed it.
This is an on-going issue which looks like a bug but i don't even know where and how to reproduce it.

Does Project contains printer settings or only Printer name to call up?

fieldOfView, 2018-03-23:
Without logs there is no way to diagnose this.