Ultimaker/Cura #2808: CR10 printhead settings

bzed, 2017-11-19:
I think the printhead settings for the CR10 series printers are still based on the default fdm profile. My measurements are:

Xmin: 30mm
Xmax: 29mm
Ymin: 32mm
Ymax: 34mm
gantry: 24mm

ChrisTerBeke, 2017-11-19:
We're not maintaining 3rd party machines ourselves, so any community pull request is welcome for this!

Ghostkeeper, 2017-11-27:
I'm not above a bit of data entry. It's probably faster than to have a pull request made, an internal ticket, to review that pull request or even than just entering the merge and push commands.

The gantry height was already set to 30, so I left that in.

closed 2017-11-27 by Ghostkeeper

bzed, 2017-11-28:
Thanks @Ghostkeeper (I'd have created a pull request at some point, just too busy to come to the point on my todo list...)

commit ab64cb57 (2017-11-27): Add print head shape

Fixes #2808.