Ultimaker/Cura #2809: Crash Start Cura 2.3.1 Windows 7

AndreaVilarinoFontan, 2017-11-19:
Hello, 
a few days ago and without having made any changes in the system, cure has stopped booting, with the following error, I have no problems with other programs, I updated the graphics card drivers, reinstalled the previous versions of cure, but the problem persists
any conclusion? Thanks

Version: 2.3.1
Platform: Windows-7-6.1.7600-SP0
Qt: 5.5.1
PyQt: 5.5.1

Exception:
Traceback (most recent call last):
  File "cura_app.py", line 61, in <module>
  File "D:\2.3\build\inst\lib\python3.4\site-packages\UM\Application.py", line 310, in getInstance
  File "D:\2.3\build\inst\lib\python3.4\site-packages\cura\CuraApplication.py", line 173, in __init__
  File "D:\2.3\build\inst\lib\python3.4\site-packages\UM\Qt\QtApplication.py", line 94, in __init__
  File "D:\2.3\build\inst\lib\python3.4\site-packages\UM\VersionUpgradeManager.py", line 107, in upgrade
  File "D:\2.3\build\inst\lib\python3.4\site-packages\UM\VersionUpgradeManager.py", line 281, in _upgradeFile
MemoryError

fieldOfView, 2017-11-19:
In c:\users\[you]\appdata\local\cura, remove cura.log. That should fix the out of memory error for now.

Upgrading to a newer version of Cura would be best, but I am guessing you run a 32 bit version of Windows

AndreaVilarinoFontan, 2017-11-20:
Thank you very much, it seems to work.

As you say I have the most updated version for 32Bits ... a shame

closed 2017-11-20 by ChrisTerBeke