Ultimaker/Cura #2812: M104 {material_print_temperature} wrong

Warbonic, 2017-11-19:
Cura 3 - recent full release
sets wrong temperature for material. The temperature that I have set for both extruders is 245 yet when my Start GCode calls for M104 {material_print_temerature} T0/T1 it sets it to 210. 
I checked my materials and it is set to a custom material. All temperature settings - Default, Printer Temp, Initial Layer, Final Printing Temp, etc. are all correct yet when I save the GCode from Cura it continually sets it to 210 instead of 245. 

This was from a fresh download and install of the current Cura version - November 19th 2017

**Here is my Start Gcode:**
G28 X0 Y0;Home X and Y only

;S = Wait for temp, R = Wait for temp even if above
M109 R150 T0 ;Sets Nozzle Temp to a temperature so bed doesn't melt when homing
M109 R150 T1
M190 S{material_bed_temperature}

G28 Z0 ;Home Z only
G29 ;Activate Auto Bed Leveling

M104 S{material_print_temperature} T0 ;sets extruders to print temperature
M104 S{material_print_temperature} T1

G92 E0
G1 F200 E3
G92 E0

**Here is the first bit of the GCode file that Cura saves:** 

;FLAVOR:Marlin
;TIME:2122
;Filament used: 0.895392m
;Layer height: 0.2
;Generated with Cura_SteamEngine 3.0.4
M82 ; absolute extrusion mode
G28 X0 Y0;Home X and Y only

;S = Wait for temp, R = Wait for temp even if above
M109 R150 T0 ;Sets Nozzle Temp to a temperature for bed
M109 R150 T1
M190 S20

G28 Z0 ;Home
G1 Z10 ;Move the platform down 15mm

G29 ;Activate Auto Bed Leveling

M104 S210 T0   **<-----------------------HERE it sets the wrong temp**
M104 S210 T1   **<-----------------------HERE it sets the wrong temp**

G92 E0
G1 F200 E3
G92 E0
;LAYER_COUNT:100
;LAYER:0
M107
M104 T1 S245  <-----------------------Sets correct temp for waiting temperature
G0 F3600 X111.756 Y137.201 Z0.1
;TYPE:SKIRT
G1 F1800 X112.198 Y136.898 E0.00891
G1 X112.681 Y136.663 E0.01784
G1 X113.192 Y136.503 E0.02675
G1 X113.722 Y136.42 E0.03567
G1 X114.009 Y136.409 E0.04045
G1 X159.991 Y136.41 E0.80513
G1 X160.526 Y136.45 E0.81405

I have no idea what is going on. I have tried {default_material_print_temperature} as well with the same results.

Warbonic, 2017-11-28:
@Ghostkeeper 
This isn't asking for a global temperature. It's asking for specific extruders. The exact same Gcode command is called by Cura a few lines later with the correct results, but it doesn't work when I input it as a start command?

This worked exactly as it should on the previous releases but since I have updated it has not.

Ghostkeeper, 2017-12-10:
It should ask for specific extruders, yes, but the `{material_print_temperature}` gets replaced with the value on the global stack. It's not actually parsing the g-code to detect that `T0` there, just blindly replacing variables with their values. That's the bug.

superflytnt, 2020-05-05:
The problem is that {material_print_temperature} on its own only reports the first extruder (T0) temp.  So your code below actually sets both T0 and T1 to the same temp (that of T0).  

M109 S{material_print_temperature} T0 ;sets extruders to print temperature
M109 S{material_print_temperature} T1

In later versions of Cura, you can specify the specific extruder, either by using an underscore or a comma.  But I'm not sure when that went into effect. This would be the correct code:

```
M109 S{material_print_temperature_0} T0 ;sets extruders to print temperature
M109 S{material_print_temperature_0} T1

```
or even

```
M109 S{material_print_temperature, 0} T0 ;sets extruders to print temperature
M109 S{material_print_temperature, 1} T1
```

fieldOfView, 2020-05-05:
> either by using an underscore or a comma

Only the comma syntax is valid. You cannot use an underscore for this.