Ultimaker/Cura #2813: STL file not loading properly

fbertram, 2017-11-19:
I am loading an STL file from Thingiverse (https://www.thingiverse.com/thing:1322352). The STL file contains 2 parts, the spinner, and the mounting plate. Cura only imports the first of the two parts, and ignores the second. Other software (Pango, Mixed Reality Viewer, 3D Builder, Paint 3D) all open the STL correctly.

[Propeller_spinner_for_Turnigy_D3530.zip](https://github.com/Ultimaker/Cura/files/1486068/Propeller_spinner_for_Turnigy_D3530.zip)

ChrisTerBeke, 2017-11-19:
I get the same issue indeed, it seems to not load the 2nd volume in a single STL file.

fgvdla, 2017-11-23:
I loaded the file into OpenSCAD and then did an export to STL. After this Cura loads correct both items.

cassidys, 2017-11-28:
I just upgraded to Cura 3.0.4 on a Mac and tried to load two STL files in at the same time, but I was having similar issues. Either one or none showed up in the solid view however it does slice them, but only one slice can be seen in layer view at a time even though it should show 5. X ray view also doesn't show up. I did have the printer plugged into USB at the time, not sure if that caused issues.

These are files I downloaded (Thingiverse and Youmagine) but I used Magics to fix up the shells and overlapping triangles so that there were no problems, just a single large closed volume shell. Saved them back as STL. 
[image]
[image]

fgvdla, 2017-11-28:
When i load a file from Thingiverse like the spinner it will indeed load one part. After loading and saving the file as STL in OpenSCAD Cura loads both parts. Every design i make in OpenSCAD Cura loads perfectly, also mutiple models at the same time. Are there different versions of STL files?

ChrisTerBeke, 2017-11-28:
Can you upload both files here? I have the feeling that the one from OpenSCAD has 1 volume defined (even though there are 1 'solids' in there), and the one from Thingiverse actually has 2 volumes in the file.

fgvdla, 2017-12-04:
I cannot upload STL files, not supported.

Ghostkeeper, 2017-12-15:
You have to zip them first before uploading on Github.

Ghostkeeper, 2017-12-15:
It seems that the file indeed has two solids in it:

    solid ascii
        facet ... <bla bla bla>
    endsolid
    solid ascii
        ...
    endsolid

We are not the only one that doesn't load that properly. When I load it in CloudCompare it also only loads one of the two solids. When I load it in OpenSCAD it works fine, like fgvdla said.

I'm thinking that this is a bug in the `numpy-stl` library we're using.

Ghostkeeper, 2017-12-15:
The leading STL format specification disallows this: http://www.fabbers.com/tech/STL_Format

Ghostkeeper, 2017-12-15:
I made a bug report at the numpy-stl repository: https://github.com/WoLpH/numpy-stl/issues/64

WoLpH, 2017-12-17:
It's actually supported but it requires a different call (since it's not returning a single mesh but multiple meshes): http://numpy-stl.readthedocs.io/en/latest/stl.html#stl.Mesh.from_multi_file

It's essentially the same as the `from_file` method but it yields multiple meshes:

```python
from stl import mesh

for m in mesh.Mesh.from_multi_file('some_file.stl'):
    pass
    # process the mesh object here
```

nallath, 2017-12-19:
As far as I understood it, the spec doesn't allow for multiple volumes.

WoLpH, 2017-12-19:
When it comes to STL there seem to be many standards unfortunately... with the ascii STL format we can safely support multiple volumes though. It doesn't transfer to the binary format though, that's single volume files only.

Ghostkeeper, 2018-01-04:
I made a fix here, thanks to the hint WoLpH gave in the numpy-stl issue: https://github.com/Ultimaker/Uranium/commit/baa1514a74c1fa7d5a8db006526112449cb0d885

It works now!

closed 2018-01-04 by Ghostkeeper