Ultimaker/Cura #2818: 3.0.4 not working with FABtotum

lahines100, 2017-11-20:
I loaded the new 3.0.4 (64) release into my FABtotum Core. When I tried to change the nozzle parameter the program quit.
[cura.log](https://github.com/Ultimaker/Cura/files/1488194/cura.log)

PaoloP74, 2017-11-22:
Uninstall Cura 3.0.4 and all the previous versions.
On Windows, remove the C:\Users\xxxx\AppData\Roaming\cura and C:\Users\xxxx\AppData\Roaming\cura
(Replace xxxx with your username)
Reinstall Cura 3.0.4.

lahines100, 2017-11-23:
In your instructions it looks as if I should remove the same file twice.

<author@noreply.github.com>
Sent: 11/22/2017 4:51:12 AM

PaoloP74, 2017-11-23:
I edit previus comment.
Subdirectory are Roaming and Local.

lahines100, 2017-11-23:
Still have a problem.
When FABtotum printer loads, program quits.
From the log, looks as if problem is with container metadata:
"2017-11-23 14:04:39,249 - WARNING - 
cura.Settings.ContainerManager.getContainerMetaDataEntry [184]: Could 
not get metadata of container  because it was not found."
Complete log is attached.

Sent: 11/23/2017 2:00:56 AM

nallath, 2017-11-27:
You didn't attach the logs. Github doesn't allow for email replies to have attachments.

lahines100, 2017-11-28:
Although I couldn't attach the logs, the problem statement was included 
in the email.

Sent: 11/27/2017 5:32:05 AM

Ghostkeeper, 2017-12-11:
The warning that you posted from the log is fairly common in normal use and doesn't result in Cura quitting. I fixed it in the `feature_local_container_server` branch which is going to be included in version 3.2. I'm thinking it must be something else because that was just a minor display issue.

closed 2018-03-02 by ChrisTerBeke

Ghostkeeper, 2018-03-26:
We haven't been able to reproduce this issue.