Ultimaker/Cura #2819: Question: Where is the code for the Raft generation?

nathanjosiah, 2017-11-20:
I'm working on a project that requires a very similar 2d mesh generation to that which is found in the raft line pattern in Cura. I have searched the codebase but I can't seem to find where the raft layers are actually generated. Can somebody point me in the right direction?

diegopradogesto, 2017-11-20:
The raft is generated in CuraEngine. You can start taking a look to our repository: https://github.com/Ultimaker/CuraEngine

Ghostkeeper, 2017-11-28:
The code to generate the areas that must be filled with raft is here: https://github.com/Ultimaker/CuraEngine/blob/master/src/raft.cpp

But what you're looking for, I think, is our infill patterns. The line pattern for rafts is actually just an infill pattern, generated in the same way that support, skin and infill is generated. So what you're actually looking for is probably the zig-zag infill pattern: https://github.com/Ultimaker/CuraEngine/blob/f8697149514f0f7cc0677b7c2b426354321c7326/src/infill.cpp#L290

closed 2018-03-09 by ianpaschal