Ultimaker/Cura #2822: Problem with master branch simulation view

smartavionics, 2017-11-21:
I'm getting this:
````
2017-11-21 10:43:17,589 - ERROR - UM.Logger.logException [76]: Exception: Import error loading module SimulationView
2017-11-21 10:43:17,589 - ERROR - UM.Logger.logException [80]: Traceback (most recent call last):
2017-11-21 10:43:17,589 - ERROR - UM.Logger.logException [80]:   File "/opt/cura/lib/python3/dist-packages/UM/PluginRegistry.py", line 472, in _findPlugin
2017-11-21 10:43:17,590 - ERROR - UM.Logger.logException [80]:     module = imp.load_module(plugin_id, file, path, desc)
2017-11-21 10:43:17,590 - ERROR - UM.Logger.logException [80]:   File "/opt/cura/lib/python3.5/imp.py", line 244, in load_module
2017-11-21 10:43:17,590 - ERROR - UM.Logger.logException [80]:     return load_package(name, filename)
2017-11-21 10:43:17,590 - ERROR - UM.Logger.logException [80]:   File "/opt/cura/lib/python3.5/imp.py", line 216, in load_package
2017-11-21 10:43:17,590 - ERROR - UM.Logger.logException [80]:     return _load(spec)
2017-11-21 10:43:17,590 - ERROR - UM.Logger.logException [80]:   File "<frozen importlib._bootstrap>", line 693, in _load
2017-11-21 10:43:17,591 - ERROR - UM.Logger.logException [80]:   File "<frozen importlib._bootstrap>", line 673, in _load_unlocked
2017-11-21 10:43:17,591 - ERROR - UM.Logger.logException [80]:   File "<frozen importlib._bootstrap_external>", line 665, in exec_module
2017-11-21 10:43:17,591 - ERROR - UM.Logger.logException [80]:   File "<frozen importlib._bootstrap>", line 222, in _call_with_frames_removed
2017-11-21 10:43:17,591 - ERROR - UM.Logger.logException [80]:   File "/opt/cura/lib/cura/plugins/SimulationView/__init__.py", line 7, in <module>
2017-11-21 10:43:17,591 - ERROR - UM.Logger.logException [80]:     from . import SimulationViewProxy, SimulationView
2017-11-21 10:43:17,592 - ERROR - UM.Logger.logException [80]:   File "/opt/cura/lib/cura/plugins/SimulationView/SimulationView.py", line 29, in <module>
2017-11-21 10:43:17,592 - ERROR - UM.Logger.logException [80]:     from .SimulationPass import SimulationPass
2017-11-21 10:43:17,592 - ERROR - UM.Logger.logException [80]:   File "/opt/cura/lib/cura/plugins/SimulationView/SimulationPass.py", line 23, in <module>
2017-11-21 10:43:17,592 - ERROR - UM.Logger.logException [80]:     from plugins.SimulationView.NozzleNode import NozzleNode
2017-11-21 10:43:17,592 - ERROR - UM.Logger.logException [80]: ImportError: No module named 'plugins'
````

fieldOfView, 2017-11-21:
Pull again, this has been fixed yesterday.
https://github.com/Ultimaker/Cura/blob/master/plugins/SimulationView/SimulationView.py#L28

smartavionics, 2017-11-21:
I already have the latest, it must be something else that's wrong.

fieldOfView, 2017-11-21:
No, you don't, because line 23 in master is not ```from plugins.SimulationView.NozzleNode import NozzleNode```

fieldOfView, 2017-11-21:
Sorry, I am looking at the wrong file.

smartavionics, 2017-11-21:
OK, I can now start master branch cura without it blowing up and I do have a simulation view in the view menu. Having sliced the model, the layer view no longer works (layers are not shown) and when I select simulation view I get a bunch of empty dialogue windows appearing and the console shows this:
````
~/3dp/Cura.git/build[mb-testing]$ 2017-11-21 14:47:19,891 - DEBUG - UM.Controller.setActiveView [85]: Setting active view to SimulationView
2017-11-21 14:47:19,898 - DEBUG - UM.Mesh.MeshData.calculateNormalsFromVertices [420]: Calculating normals took 0.0012772083282470703 seconds
2017-11-21 14:47:19,937 - DEBUG - UM.Mesh.MeshData.approximateConvexHull [387]: approximateConvexHull(target_count=1024) Calculating 3D convex hull took 0.0377202033996582 seconds. 351 input vertices. 351 output vertices.
2017-11-21 14:47:19,939 - DEBUG - STLReader.STLReader.read [79]: Loaded a mesh with 12612 vertices
2017-11-21 14:47:19,947 - CRITICAL - cura.CrashHandler.__init__ [62]: An uncaught exception has occurred!
2017-11-21 14:47:19,947 - CRITICAL - cura.CrashHandler.__init__ [65]: TypeError: unable to convert a Python 'NoneType' object to a C++ 'bool' instance
2017-11-21 14:47:19,950 - CRITICAL - cura.CrashHandler.__init__ [62]: An uncaught exception has occurred!
2017-11-21 14:47:19,951 - CRITICAL - cura.CrashHandler.__init__ [65]: TypeError: unable to convert a Python 'NoneType' object to a C++ 'bool' instance
2017-11-21 14:47:19,953 - DEBUG - UM.View.GL.ShaderProgram.load [55]: Loading shader file [/opt/cura/lib/cura/plugins/SimulationView/simulationview_composite.shader]...
2017-11-21 14:47:20,967 - DEBUG - CuraEngineBackend.ProcessSlicedLayersJob.run [233]: Processing layers took 0.9732871055603027 seconds
2017-11-21 14:47:20,992 - CRITICAL - cura.CrashHandler.__init__ [62]: An uncaught exception has occurred!
2017-11-21 14:47:20,993 - CRITICAL - cura.CrashHandler.__init__ [65]: TypeError: unable to convert a Python 'NoneType' object to a C++ 'bool' instance
2017-11-21 14:47:20,999 - CRITICAL - cura.CrashHandler.__init__ [62]: An uncaught exception has occurred!
2017-11-21 14:47:21,000 - CRITICAL - cura.CrashHandler.__init__ [65]: TypeError: unable to convert a Python 'NoneType' object to a C++ 'int' instance
2017-11-21 14:47:21,000 - CRITICAL - cura.CrashHandler.__init__ [62]: An uncaught exception has occurred!
2017-11-21 14:47:21,001 - CRITICAL - cura.CrashHandler.__init__ [65]: TypeError: unable to convert a Python 'NoneType' object to a C++ 'int' instance
2017-11-21 14:47:21,001 - CRITICAL - cura.CrashHandler.__init__ [62]: An uncaught exception has occurred!
2017-11-21 14:47:21,001 - CRITICAL - cura.CrashHandler.__init__ [65]: TypeError: unable to convert a Python 'NoneType' object to a C++ 'int' instance
2017-11-21 14:47:21,002 - CRITICAL - cura.CrashHandler.__init__ [62]: An uncaught exception has occurred!
2017-11-21 14:47:21,002 - CRITICAL - cura.CrashHandler.__init__ [65]: TypeError: unable to convert a Python 'NoneType' object to a C++ 'int' instance
2017-11-21 14:47:21,003 - CRITICAL - cura.CrashHandler.__init__ [62]: An uncaught exception has occurred!
2017-11-21 14:47:21,003 - CRITICAL - cura.CrashHandler.__init__ [65]: TypeError: unable to convert a Python 'NoneType' object to a C++ 'int' instance
2017-11-21 14:47:21,004 - CRITICAL - cura.CrashHandler.__init__ [62]: An uncaught exception has occurred!
2017-11-21 14:47:21,004 - CRITICAL - cura.CrashHandler.__init__ [65]: TypeError: unable to convert a Python 'NoneType' object to a C++ 'int' instance
2017-11-21 14:47:21,006 - CRITICAL - cura.CrashHandler.__init__ [62]: An uncaught exception has occurred!
2017-11-21 14:47:21,007 - CRITICAL - cura.CrashHandler.__init__ [65]: TypeError: unable to convert a Python 'NoneType' object to a C++ 'int' instance
2017-11-21 14:47:21,008 - CRITICAL - cura.CrashHandler.__init__ [62]: An uncaught exception has occurred!
2017-11-21 14:47:21,008 - CRITICAL - cura.CrashHandler.__init__ [65]: TypeError: unable to convert a Python 'NoneType' object to a C++ 'int' instance
2017-11-21 14:47:21,009 - CRITICAL - cura.CrashHandler.__init__ [62]: An uncaught exception has occurred!
2017-11-21 14:47:21,009 - CRITICAL - cura.CrashHandler.__init__ [65]: TypeError: unable to convert a Python 'NoneType' object to a C++ 'bool' instance
2017-11-21 14:47:21,009 - CRITICAL - cura.CrashHandler.__init__ [62]: An uncaught exception has occurred!
2017-11-21 14:47:21,010 - CRITICAL - cura.CrashHandler.__init__ [65]: TypeError: unable to convert a Python 'NoneType' object to a C++ 'int' instance
2017-11-21 14:47:21,010 - CRITICAL - cura.CrashHandler.__init__ [62]: An uncaught exception has occurred!
2017-11-21 14:47:21,010 - CRITICAL - cura.CrashHandler.__init__ [65]: TypeError: unable to convert a Python 'NoneType' object to a C++ 'int' instance
2017-11-21 14:47:21,010 - CRITICAL - cura.CrashHandler.__init__ [62]: An uncaught exception has occurred!
2017-11-21 14:47:21,011 - CRITICAL - cura.CrashHandler.__init__ [65]: TypeError: unable to convert a Python 'NoneType' object to a C++ 'int' instance
2017-11-21 14:47:21,011 - CRITICAL - cura.CrashHandler.__init__ [62]: An uncaught exception has occurred!
2017-11-21 14:47:21,011 - CRITICAL - cura.CrashHandler.__init__ [65]: TypeError: unable to convert a Python 'NoneType' object to a C++ 'int' instance
2017-11-21 14:47:21,012 - CRITICAL - cura.CrashHandler.__init__ [62]: An uncaught exception has occurred!
[... cut ...]
2017-11-21 14:47:21,036 - DEBUG - UM.View.GL.ShaderProgram.load [55]: Loading shader file [/opt/cura/lib/cura/plugins/SimulationView/layers3d.shader]...
2017-11-21 14:47:21,238 - DEBUG - UM.View.GL.ShaderProgram.load [55]: Loading shader file [/opt/cura/lib/cura/plugins/SimulationView/layers3d_shadow.shader]...
2017-11-21 14:47:21,395 - DEBUG - UM.View.GL.ShaderProgram.load [55]: Loading shader file [/opt/cura/share/uranium/resources/shaders/toolhandle.shader]...
2017-11-21 14:47:21,401 - DEBUG - UM.View.GL.ShaderProgram.load [55]: Loading shader file [/opt/cura/share/uranium/resources/shaders/color.shader]...
````

diegopradogesto, 2017-11-21:
Yesterday we have some issues with the plugin and we needed to fix them. That's the reason why is not properly working now on master but it will be fixed as soon as possible.

smartavionics, 2017-11-21:
OK, actually, I just did a cmake .. before installing cura and I think I was possibly missing something before because I tried again and now I am seeing a simulation view. I will post again if it's still a problem

smartavionics, 2017-11-21:
No, I spoke too soon because I switched from the simulation view to the layer view and then back to the simulation view and it crashed just like it did before
````
2017-11-21 14:54:23,367 - DEBUG - UM.Controller.setActiveView [85]: Setting active view to SimulationView
2017-11-21 14:54:23,379 - CRITICAL - cura.CrashHandler.__init__ [62]: An uncaught exception has occurred!
2017-11-21 14:54:23,380 - CRITICAL - cura.CrashHandler.__init__ [65]: TypeError: unable to convert a Python 'NoneType' object to a C++ 'bool' instance
2017-11-21 14:54:23,384 - CRITICAL - cura.CrashHandler.__init__ [62]: An uncaught exception has occurred!
2017-11-21 14:54:23,385 - CRITICAL - cura.CrashHandler.__init__ [65]: TypeError: unable to convert a Python 'NoneType' object to a C++ 'bool' instance
````

smartavionics, 2017-11-21:
It's still unreliable because sometimes I can get a simulation view and other times it blows up like above. But when it works, it looks good. I would still like to see the info for the current move (rate, distance, z, flow, etc.) being displayed somewhere but it's already useful as it is.

smartavionics, 2017-11-22:
Here's a thought. As mentioned elsewhere, it would be very useful to see info about the current line/move just like craftware has. I realise that not everyone would want to see that and so it would be good to be able to enable/disable the info display. My thought was that to save having yet another checkbox in the GUI, why not just bind a key such as 'i' (for info) to the line slider. That already has bindings for fwd/back movement so it must be trivial to add a binding to toggle the display of the info?

smartavionics, 2017-11-23:
Hi, I'm reusing this issue for a new problem I have just discovered. The following image shows a sliced model that's come out all black. I know cats are often black but it should be coloured based on the feedrate. Furthermore, the displayed feedrate values are obviously screwed. My guess is that somewhere the colouring code is getting either a very high or low feedrate value and that makes the displayed range wrong.

[image]
This appears to be a model specific problem because other models sliced with the same profile look OK.

fieldOfView, 2017-11-23:
> This appears to be a model specific problem 

Got link?

The shader seems to "crash" because it can't determine where the current value lies between the min and max that it is given. That would suggest the error is in determining the min and max value.

smartavionics, 2017-11-23:
Thingyverse low poly cat

smartavionics, 2017-11-23:
Interestingly, if I repeatedly reslice (using the same parameters), sometimes the simulation view looks OK (cat is blue!) and sometimes the error occurs and the cat is black.

smartavionics, 2017-11-23:
Another problem with the feedrate colour scheme is that because the travel rate I use (300mm/S) is much larger than the rates used when actually extruding (normally < 50mm/S), there is almost no difference in the colours used for the different extrusions (i.e. they are all blue). I think to be useful, the colouring needs to be scaled so that the different print speeds can be identified more easily. I suggest that the blue to red range of colours is only used for non-travel moves and that travel moves have some easily visible colour (i.e. bright yellow). Alternatively, you could use a non-linear scaling so that the majority of the colours represent low speed print/travel moves.
[image]

diegopradogesto, 2017-11-23:
Regarding the black cat thing, the feedrate seems to be calculated because the gradient shows the minimum and maximum (although not trimming the decimals). If you rotate the view and Cura renders again, is it still happening? I have seen it before in Linux, is that correct?

Regarding the gradient colors, I got the same feedback from colleagues within the company. So what we are doing is changing the gradients to be more aware of the smaller values. However in my opinion we still need to show the travel feedrate.

Thanks for the feedback!

smartavionics, 2017-11-23:
> Regarding the black cat thing, the feedrate seems to be calculated because the gradient shows the minimum and maximum (although not trimming the decimals). If you rotate the view and Cura renders again, is it still happening? I have seen it before in Linux, is that correct?

Rotating the view doesn't appear to make any difference but if I reslice using the same settings over and over sometimes the cat is blue and sometimes it is black.

smartavionics, 2017-11-27:
Any luck with a fix for the black cat problem? The current master branch still exhibits this behaviour.

diegopradogesto, 2017-11-27:
I tried but I got no error. Maybe is setting specific. Could you send me the project file so I can take a look?

smartavionics, 2017-11-27:
I initially had the profile set to Fine and it wouldn't fail on repeated reslicing. I then changed to Fast (0.2mm layers) and it failed first slice, thereafter it sometimes fails and sometimes not.

[black_cat_project.zip](https://github.com/Ultimaker/Cura/files/1506337/black_cat_project.zip)

smartavionics, 2017-11-27:
Haven't yet got it to fail on 0.1mm layers but with 0.2mm layers it fails roughly 50% of the time. Methinks this is one of those special Schrodingers cats that exist/don't exist at the same time.

diegopradogesto, 2017-11-28:
I still can not reproduce it, sorry. Could it be related with graphic drivers?

smartavionics, 2017-11-28:
Could be, do you have anyone there running Linux on a machine that's using Intel graphics (i3, i5, etc.)? If so, see if the problem occurs on their systems.

diegopradogesto, 2017-11-28:
Yes, I already ask my colleagues using Linux with Intel graphic card and we can't reproduce it. If it just happens with the cat, then could be of course the case of the Schrodinger's cat XDDD.
Maybe it was unexpectedly fixed with the changes we are doing.

smartavionics, 2017-11-28:
No, it happens with every model when I use 0.2mm layer height. I am not seeing any other graphics problems in cura or any other program so I don't know whether it really is related to the graphics driver. I am currently using:
````
2017-11-28 13:19:57,870 - DEBUG - UM.View.GL.OpenGL.__init__ [93]: OpenGL Version:  4.5 (Core Profile) Mesa 17.0.7
2017-11-28 13:19:57,870 - DEBUG - UM.View.GL.OpenGL.__init__ [94]: OpenGL Vendor:   Intel Open Source Technology Center
2017-11-28 13:19:57,870 - DEBUG - UM.View.GL.OpenGL.__init__ [95]: OpenGL Renderer: Mesa DRI Intel(R) HD Graphics 520 (Skylake GT2) 
````

smartavionics, 2017-11-28:
My current guess is that there is a race occurring. I think the renderer sometimes gets invoked before the values of _max_feedrate, etc. have been calculated. Sorry, I don't understand enough of how this stuff works to really be able to help much in finding what is going wrong here but I shall continue to explore.
How do the calls to resetLayerData(), _onSceneChanged() and the call to the renderer get synchronised?

diegopradogesto, 2017-11-30:
If you wait for the slicing process staying at the SolidView and once it finishes you change to the LayerView, it still happens? Because the feedrate information is populated after the slicing process and when switching to LayerView is when the render process starts.
I assume that you are using your own compiled CuraEngine. Does it happens if you run the precompiled package from Ultimaker's website?

smartavionics, 2017-11-30:
Sorry, I have given up trying to work out what's wrong here. I'm sure I won't be the only one that has this problem so when others start complaining maybe it will become more obvious where the problem lies.

closed 2017-11-30 by smartavionics