Ultimaker/Cura #2824: Cura 3.0.4 always bumps print during bridging of 2 separate pieces...

rickmjohnston61, 2017-11-22:
Cura 3.0.4 always bumps print during bridging of 2 separate pieces? I guess I cannot say Cura but I cannot make any adjustments on printer except x/y???

I've adjusted everything with my UM 3 printer and nothing. I try to brace and still it hits the piece too hard and breaks it from print bed? BTW: it always auto-adjust and until now having gotten superior prints.

Really need Z offset setting like is supposedly required for printing with nylon but this is simply PLA...and I see no such setting. I tried manually adjusting and this still does not work!

Any thoughts PLEASE!

Rick...

diegopradogesto, 2017-11-24:
Have you take a look at our setting "Z Hop Height"? Maybe you can tweak the values to avoid those bumps.

rickmjohnston61, 2017-11-24:
Thanks.

Where is that setting? I'm not finding it in cura settings?

Cheers!

Rick…

Sent from my mobile phone... please excuse brevity and typos...

Ghostkeeper, 2017-12-01:
Enable "Z Hop When Retracted" first. This will enable a feature that causes the nozzle to go up (or the bed down) when it moves from one part to another.

When that is enabled, the Z Hop Height setting will appear to adjust how high it hops (it hops 2mm by default).

rickmjohnston61, 2017-12-01:
Thank you I had found this in Cura 3.04 Preferences-Configure Cura->Settings->Travel and can check on (setting were all unchecked by default). I then under Materials->Selected Material->Extruder 1/2 Printer Settings it lists 1.5mm which I cannot change.

Anyway thanks I can go from here and so far my issue seemed to be isolated as I have gotten good prints once again without any issues.

closed 2017-12-01 by rickmjohnston61