Ultimaker/Cura #2828: CRITICAL - cura.CrashHandler.__init__ [62]: An uncaught exception has occurred

marcosfg, 2017-11-22:
I'm having trouble running the master branch in my OSX computer.

When I run the cura_app.py I always get this error:

```
2017-11-22 16:17:09,757 - CRITICAL - cura.CrashHandler.__init__ [62]: An uncaught exception has occurred!
2017-11-22 16:17:09,765 - CRITICAL - cura.CrashHandler.__init__ [65]: Traceback (most recent call last):
2017-11-22 16:17:09,766 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/Applications/PyCharm.app/Contents/helpers/pydev/pydevd.py", line 1599, in <module>
2017-11-22 16:17:09,767 - CRITICAL - cura.CrashHandler.__init__ [65]:     globals = debugger.run(setup['file'], None, None, is_module)
2017-11-22 16:17:09,768 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/Applications/PyCharm.app/Contents/helpers/pydev/pydevd.py", line 1026, in run
2017-11-22 16:17:09,768 - CRITICAL - cura.CrashHandler.__init__ [65]:     pydev_imports.execfile(file, globals, locals)  # execute the script
2017-11-22 16:17:09,769 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/Applications/PyCharm.app/Contents/helpers/pydev/_pydev_imps/_pydev_execfile.py", line 18, in execfile
2017-11-22 16:17:09,770 - CRITICAL - cura.CrashHandler.__init__ [65]:     exec(compile(contents+"\n", file, 'exec'), glob, loc)
2017-11-22 16:17:09,771 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/Users/mgomes/git/Ultimaker/Cura/cura_app.py", line 82, in <module>
2017-11-22 16:17:09,771 - CRITICAL - cura.CrashHandler.__init__ [65]:     app = cura.CuraApplication.CuraApplication.getInstance()
2017-11-22 16:17:09,772 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/usr/local/lib/python3.6/site-packages/UM/Application.py", line 321, in getInstance
2017-11-22 16:17:09,773 - CRITICAL - cura.CrashHandler.__init__ [65]:     Application._instance = cls()
2017-11-22 16:17:09,773 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/Users/mgomes/git/Ultimaker/Cura/cura/CuraApplication.py", line 287, in __init__
2017-11-22 16:17:09,774 - CRITICAL - cura.CrashHandler.__init__ [65]:     ContainerRegistry.getInstance().load()
2017-11-22 16:17:09,775 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/Users/mgomes/git/Ultimaker/Cura/cura/Settings/CuraContainerRegistry.py", line 260, in load
2017-11-22 16:17:09,776 - CRITICAL - cura.CrashHandler.__init__ [65]:     self._registerSingleExtrusionMachinesExtruderStacks()
2017-11-22 16:17:09,776 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/Users/mgomes/git/Ultimaker/Cura/cura/Settings/CuraContainerRegistry.py", line 403, in _registerSingleExtrusionMachinesExtruderStacks
2017-11-22 16:17:09,777 - CRITICAL - cura.CrashHandler.__init__ [65]:     machines = ContainerRegistry.getInstance().findContainerStacks(machine_extruder_trains = {"0": "fdmextruder"})
2017-11-22 16:17:09,778 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/usr/local/lib/python3.6/site-packages/UM/Settings/ContainerRegistry.py", line 95, in findContainerStacks
2017-11-22 16:17:09,778 - CRITICAL - cura.CrashHandler.__init__ [65]:     return cast(List[ContainerStack], self.findContainers(ContainerStack, **kwargs))
2017-11-22 16:17:09,779 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/usr/local/lib/python3.6/site-packages/UM/Settings/ContainerRegistry.py", line 125, in findContainers
2017-11-22 16:17:09,780 - CRITICAL - cura.CrashHandler.__init__ [65]:     if query in self._query_cache:
2017-11-22 16:17:09,781 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/usr/local/lib/python3.6/site-packages/UM/Settings/ContainerQuery.py", line 80, in __hash__
2017-11-22 16:17:09,781 - CRITICAL - cura.CrashHandler.__init__ [65]:     return hash(self.__key())
2017-11-22 16:17:09,782 - CRITICAL - cura.CrashHandler.__init__ [65]: TypeError: unhashable type: 'dict'

```

Any help?

ChrisTerBeke, 2017-11-22:
Did you update Uranium as well?

marcosfg, 2017-11-23:
Thank you Chris, after updating Uranium it actually proceeded, however I ended with another problem...

After that I had a problem where the main window did not show. I found it was related to pyqt5, where the 5.9.2 version is not compatible.

Now I managed to have at least the basic functionality from cura working.

Thanks

ChrisTerBeke, 2017-11-23:
5.9.2 is indeed not working properly. try 5.8.x, that's the version we also use in our builds.

Ghostkeeper, 2017-11-28:
Qt 5.9.3 released around the time you made this issue. That should fix the crash you were getting (and I'm a bit curious if it actually does fix it).

closed 2017-11-28 by Ghostkeeper