Ultimaker/Cura #2830: Fatal Python error: Segmentation fault

sfornengo, 2017-11-22:
from ubuntu cura [master] ppa.
no longer works with the last update.

$ cura
UM.Resources.__initializeStoragePaths [342]: Initializing storage paths
UM.Resources.__initializeStoragePaths [353]: Config storage path is /home/stf/.config/cura/3.0
UM.Resources.__initializeStoragePaths [361]: Data storage path is /home/stf/.local/share/cura/3.0
UM.Resources.__initializeStoragePaths [373]: Cache storage path is /home/stf/.cache/cura/3.0
UM.View.GL.OpenGLContext.detectBestOpenGLVersion [103]: Trying OpenGL context 4.1...
UM.View.GL.OpenGLContext.detectBestOpenGLVersion [113]: Yay, we got at least OpenGL 4.1 core: 4.1 Core profile
UM.Qt.QtApplication.__init__ [82]: Detected most suitable OpenGL context version: 4.1 Core profile
UM.Qt.QtApplication.__init__ [98]: Command line arguments: {'external-backend': False, 'disable-textures': False, 'file': [], 'headless': False, 'qmljsdebugger': None, 'single_instance': False}
2017-11-22 19:45:31,116 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin ConsoleLogger
2017-11-22 19:45:31,128 - INFO - CuraEngineBackend.CuraEngineBackend.__init__ [64]: Found CuraEngine at: /usr/bin/CuraEngine
2017-11-22 19:45:31,129 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin CuraEngineBackend
2017-11-22 19:45:31,137 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin TranslateTool
2017-11-22 19:45:31,142 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin MirrorTool
2017-11-22 19:45:31,159 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin RotateTool
2017-11-22 19:45:31,167 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin CameraTool
2017-11-22 19:45:31,171 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin SelectionTool
2017-11-22 19:45:31,179 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin ScaleTool
2017-11-22 19:45:31,185 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin LocalFileOutputDevice
2017-11-22 19:45:31,188 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin FileLogger
2017-11-22 19:45:31,188 - WARNING - UM.PluginRegistry.loadPlugin [200]: Plugin ConsoleLogger was already loaded
2017-11-22 19:45:31,190 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin SimpleView
2017-11-22 19:45:31,194 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin OBJWriter
2017-11-22 19:45:31,198 - WARNING - STLReader.STLReader.<module> [25]: Could not find numpy-stl, falling back to slower code.
2017-11-22 19:45:31,200 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin STLReader
2017-11-22 19:45:31,209 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin STLWriter
2017-11-22 19:45:31,218 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin OBJReader
2017-11-22 19:45:31,224 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin OrientationPlugin
2017-11-22 19:45:31,227 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin VersionUpgrade22to24
2017-11-22 19:45:31,229 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin VersionUpgrade25to26
2017-11-22 19:45:31,231 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin VersionUpgrade27to30
2017-11-22 19:45:31,235 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin VersionUpgrade21to22
2017-11-22 19:45:31,237 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin VersionUpgrade26to27
2017-11-22 19:45:31,239 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin VersionUpgrade30to31
2017-11-22 19:45:31,243 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin ChangeLogPlugin
2017-11-22 19:45:31,266 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin USBPrinting
2017-11-22 19:45:31,371 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin UM3NetworkPrinting
2017-11-22 19:45:31,388 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin 3MFReader
2017-11-22 19:45:31,392 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin UserAgreementPlugin
2017-11-22 19:45:31,393 - WARNING - UM.PluginRegistry.loadPlugin [200]: Plugin OrientationPlugin was already loaded
2017-11-22 19:45:31,411 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin X3DReader
2017-11-22 19:45:31,420 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin XmlMaterialProfile
2017-11-22 19:45:31,427 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin CuraProfileWriter
2017-11-22 19:45:31,455 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin OctoPrintPlugin
2017-11-22 19:45:31,464 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin XRayView
2017-11-22 19:45:31,470 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin RemovableDriveOutputDevice
2017-11-22 19:45:31,475 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin PerObjectSettingsTool
2017-11-22 19:45:31,479 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin LegacyProfileReader
2017-11-22 19:45:31,479 - WARNING - UM.PluginRegistry.loadPlugin [200]: Plugin CuraEngineBackend was already loaded
2017-11-22 19:45:31,483 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin MachineSettingsAction
2017-11-22 19:45:31,513 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin SliceInfoPlugin
2017-11-22 19:45:31,528 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin PluginBrowser
2017-11-22 19:45:31,539 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin 3MFWriter
2017-11-22 19:45:31,545 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin FirmwareUpdateChecker
2017-11-22 19:45:31,557 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin UltimakerMachineActions
2017-11-22 19:45:31,561 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin GCodeProfileReader
2017-11-22 19:45:31,565 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin ImageReader
2017-11-22 19:45:31,574 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin GCodeReader
2017-11-22 19:45:31,577 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin CuraProfileReader
2017-11-22 19:45:31,584 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin LayerView
2017-11-22 19:45:31,590 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin PostProcessingPlugin
2017-11-22 19:45:31,595 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin SolidView
2017-11-22 19:45:31,598 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin AutoSave
2017-11-22 19:45:31,602 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin GCodeWriter
2017-11-22 19:45:31,616 - INFO - UM.VersionUpgradeManager.upgrade [112]: Looking for old configuration files to upgrade.
2017-11-22 19:45:31,635 - DEBUG - USBPrinting.USBPrinterOutputDevice._connect [312]: Attempting to connect to /dev/ttyACM0
2017-11-22 19:45:31,996 - DEBUG - UM.Backend.Backend._logSocketState [174]: Socket state changed to Listening
[... 253 lines cut ...]
Thread 0x00007efdcaffd700 (most recent call first):
  File "/usr/lib/python3/dist-packages/zeroconf.py", line 829 in run
  File "/usr/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/usr/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007efdcb7fe700 (most recent call first):
  File "/usr/lib/python3.5/threading.py", line 297 in wait
  File "/usr/lib/python3.5/threading.py", line 549 in wait
  File "/usr/lib/cura/plugins/UM3NetworkPrinting/NetworkPrinterOutputDevicePlugin.py", line 326 in _handleOnServiceChangedRequests
  File "/usr/lib/python3.5/threading.py", line 862 in run
  File "/usr/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/usr/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007efdd1053700 (most recent call first):
  File "/usr/lib/cura/plugins/USBPrinting/USBPrinterOutputDeviceManager.py", line 88 in _updateThread
  File "/usr/lib/python3.5/threading.py", line 862 in run
  File "/usr/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/usr/lib/python3.5/threading.py", line 882 in _bootstrap

Current thread 0x00007efe1478c700 (most recent call first):
  File "/usr/lib/python3/dist-packages/UM/Qt/ListModel.py", line 62 in setItems
  File "/usr/lib/python3/dist-packages/cura/Settings/ExtrudersModel.py", line 236 in __updateExtruders
  File "/usr/lib/python3/dist-packages/cura/CuraApplication.py", line 731 in run
  File "/usr/bin/cura", line 83 in <module>
Erreur de segmentation (core dumped)

diegopradogesto, 2017-11-24:
The segmentation fault seems to be raising in the OctoPrintPlugin that is supported by @fieldOfView .
Are you sure you have the latest build?

sfornengo, 2017-11-24:
Yes, the lastest build from https://launchpad.net/~thopiekar/+archive/ubuntu/cura-master
The problem came from an update few days ago, all worked great before.
Removing OctoPrintPlugin and other plugins lead to the same segmentation fault:

$ cura
UM.Resources.__initializeStoragePaths [342]: Initializing storage paths
UM.Resources.__initializeStoragePaths [353]: Config storage path is /home/stf/.config/cura/3.0
UM.Resources.__initializeStoragePaths [361]: Data storage path is /home/stf/.local/share/cura/3.0
UM.Resources.__initializeStoragePaths [373]: Cache storage path is /home/stf/.cache/cura/3.0
UM.View.GL.OpenGLContext.detectBestOpenGLVersion [103]: Trying OpenGL context 4.1...
UM.View.GL.OpenGLContext.detectBestOpenGLVersion [113]: Yay, we got at least OpenGL 4.1 core: 4.1 Core profile
UM.Qt.QtApplication.__init__ [82]: Detected most suitable OpenGL context version: 4.1 Core profile
UM.Qt.QtApplication.__init__ [98]: Command line arguments: {'single_instance': False, 'qmljsdebugger': None, 'disable-textures': False, 'file': [], 'headless': False, 'external-backend': False}
2017-11-24 09:45:02,181 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin ConsoleLogger
2017-11-24 09:45:02,211 - INFO - CuraEngineBackend.CuraEngineBackend.__init__ [64]: Found CuraEngine at: /usr/bin/CuraEngine
2017-11-24 09:45:02,212 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin CuraEngineBackend
2017-11-24 09:45:02,220 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin TranslateTool
2017-11-24 09:45:02,226 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin MirrorTool
2017-11-24 09:45:02,237 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin RotateTool
2017-11-24 09:45:02,251 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin CameraTool
2017-11-24 09:45:02,255 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin SelectionTool
2017-11-24 09:45:02,264 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin ScaleTool
2017-11-24 09:45:02,270 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin LocalFileOutputDevice
2017-11-24 09:45:02,280 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin FileLogger
2017-11-24 09:45:02,282 - WARNING - UM.PluginRegistry.loadPlugin [200]: Plugin ConsoleLogger was already loaded
2017-11-24 09:45:02,290 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin SimpleView
2017-11-24 09:45:02,293 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin OBJWriter
2017-11-24 09:45:02,298 - WARNING - STLReader.STLReader.<module> [25]: Could not find numpy-stl, falling back to slower code.
2017-11-24 09:45:02,300 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin STLReader
2017-11-24 09:45:02,304 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin STLWriter
2017-11-24 09:45:02,307 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin OBJReader
2017-11-24 09:45:02,316 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin OrientationPlugin
2017-11-24 09:45:02,326 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin VersionUpgrade22to24
2017-11-24 09:45:02,330 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin VersionUpgrade25to26
2017-11-24 09:45:02,332 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin VersionUpgrade27to30
2017-11-24 09:45:02,336 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin VersionUpgrade21to22
2017-11-24 09:45:02,339 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin VersionUpgrade26to27
2017-11-24 09:45:02,341 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin VersionUpgrade30to31
2017-11-24 09:45:02,344 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin ChangeLogPlugin
2017-11-24 09:45:02,382 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin USBPrinting
2017-11-24 09:45:02,511 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin UM3NetworkPrinting
2017-11-24 09:45:02,528 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin 3MFReader
2017-11-24 09:45:02,538 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin UserAgreementPlugin
2017-11-24 09:45:02,560 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin X3DReader
2017-11-24 09:45:02,567 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin XmlMaterialProfile
2017-11-24 09:45:02,573 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin CuraProfileWriter
2017-11-24 09:45:02,578 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin XRayView
2017-11-24 09:45:02,588 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin RemovableDriveOutputDevice
2017-11-24 09:45:02,597 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin PerObjectSettingsTool
2017-11-24 09:45:02,605 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin LegacyProfileReader
2017-11-24 09:45:02,606 - WARNING - UM.PluginRegistry.loadPlugin [200]: Plugin CuraEngineBackend was already loaded
2017-11-24 09:45:02,615 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin MachineSettingsAction
2017-11-24 09:45:02,660 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin SliceInfoPlugin
2017-11-24 09:45:02,679 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin PluginBrowser
2017-11-24 09:45:02,685 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin 3MFWriter
2017-11-24 09:45:02,691 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin FirmwareUpdateChecker
[... cut ...]
  File "/usr/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/usr/lib/python3.5/threading.py", line 882 in _bootstrap

Current thread 0x00007f9d758e6700 (most recent call first):
  File "/usr/lib/python3/dist-packages/UM/Qt/ListModel.py", line 62 in setItems
  File "/usr/lib/python3/dist-packages/cura/Settings/ExtrudersModel.py", line 236 in __updateExtruders
  File "/usr/lib/python3/dist-packages/cura/CuraApplication.py", line 731 in run
  File "/usr/bin/cura", line 83 in <module>
Erreur de segmentation (core dumped)

fieldOfView, 2017-11-24:
As far as I can tell, the crash has nothing to do with OctoPrintPlugin (it just happened to be the last piece of code that logged something)

diegopradogesto, 2017-11-24:
OK. This can be related with some changes that we are doing in the stacks. We have an idea about what is going on, although I couldn't reproduce the segfault. 
In the meanwhile, could you try to clean up the configuration and see if the error is still there?

sfornengo, 2017-11-24:
I tried uninstalling and reinstalling, no change.
I tried to remove .cache/cura, no change.
I tried to remove .config/cura and now the interface opened before the app crash, with this new error message:
$ cura
UM.Resources.__initializeStoragePaths [342]: Initializing storage paths
UM.Resources.__initializeStoragePaths [353]: Config storage path is /home/stf/.config/cura/3.0
UM.Resources.__initializeStoragePaths [361]: Data storage path is /home/stf/.local/share/cura/3.0
UM.Resources.__initializeStoragePaths [373]: Cache storage path is /home/stf/.cache/cura/3.0
UM.View.GL.OpenGLContext.detectBestOpenGLVersion [103]: Trying OpenGL context 4.1...
UM.View.GL.OpenGLContext.detectBestOpenGLVersion [113]: Yay, we got at least OpenGL 4.1 core: 4.1 Core profile
UM.Qt.QtApplication.__init__ [82]: Detected most suitable OpenGL context version: 4.1 Core profile
UM.Qt.QtApplication.__init__ [98]: Command line arguments: {'external-backend': False, 'qmljsdebugger': None, 'file': [], 'single_instance': False, 'disable-textures': False, 'headless': False}
2017-11-24 11:39:35,936 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin ConsoleLogger
2017-11-24 11:39:35,947 - INFO - CuraEngineBackend.CuraEngineBackend.__init__ [64]: Found CuraEngine at: /usr/bin/CuraEngine
2017-11-24 11:39:35,948 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin CuraEngineBackend
2017-11-24 11:39:35,956 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin TranslateTool
2017-11-24 11:39:35,961 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin MirrorTool
2017-11-24 11:39:35,968 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin RotateTool
2017-11-24 11:39:35,973 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin CameraTool
2017-11-24 11:39:35,976 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin SelectionTool
2017-11-24 11:39:35,984 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin ScaleTool
2017-11-24 11:39:35,990 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin LocalFileOutputDevice
2017-11-24 11:39:35,993 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin FileLogger
2017-11-24 11:39:35,993 - WARNING - UM.PluginRegistry.loadPlugin [200]: Plugin ConsoleLogger was already loaded
2017-11-24 11:39:35,996 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin SimpleView
2017-11-24 11:39:35,999 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin OBJWriter
2017-11-24 11:39:36,004 - WARNING - STLReader.STLReader.<module> [25]: Could not find numpy-stl, falling back to slower code.
2017-11-24 11:39:36,006 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin STLReader
2017-11-24 11:39:36,010 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin STLWriter
2017-11-24 11:39:36,012 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin OBJReader
2017-11-24 11:39:36,016 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin OrientationPlugin
2017-11-24 11:39:36,018 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin VersionUpgrade22to24
2017-11-24 11:39:36,021 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin VersionUpgrade25to26
2017-11-24 11:39:36,023 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin VersionUpgrade27to30
2017-11-24 11:39:36,026 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin VersionUpgrade21to22
2017-11-24 11:39:36,029 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin VersionUpgrade26to27
2017-11-24 11:39:36,031 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin VersionUpgrade30to31
2017-11-24 11:39:36,034 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin ChangeLogPlugin
2017-11-24 11:39:36,057 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin USBPrinting
2017-11-24 11:39:36,148 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin UM3NetworkPrinting
2017-11-24 11:39:36,175 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin 3MFReader
2017-11-24 11:39:36,179 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin UserAgreementPlugin
2017-11-24 11:39:36,192 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin X3DReader
2017-11-24 11:39:36,197 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin XmlMaterialProfile
2017-11-24 11:39:36,200 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin CuraProfileWriter
2017-11-24 11:39:36,203 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin XRayView
2017-11-24 11:39:36,208 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin RemovableDriveOutputDevice
2017-11-24 11:39:36,212 - WARNING - UM.Preferences.getValue [81]: Tried to get the value of non-existing setting cura/active_mode.
2017-11-24 11:39:36,213 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin PerObjectSettingsTool
2017-11-24 11:39:36,216 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin LegacyProfileReader
2017-11-24 11:39:36,217 - WARNING - UM.PluginRegistry.loadPlugin [200]: Plugin CuraEngineBackend was already loaded
2017-11-24 11:39:36,221 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin MachineSettingsAction
2017-11-24 11:39:36,252 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin SliceInfoPlugin
2017-11-24 11:39:36,263 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin PluginBrowser
2017-11-24 11:39:36,269 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin 3MFWriter
[... cut ...]
2017-11-24 11:40:31,070 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/usr/lib/python3/dist-packages/cura/CuraApplication.py", line 1412, in _readMeshFinished
2017-11-24 11:40:31,071 - CRITICAL - cura.CrashHandler.__init__ [65]:     if node.getBoundingBox().width < self._volume.getBoundingBox().width or node.getBoundingBox().depth < self._volume.getBoundingBox().depth:
2017-11-24 11:40:31,072 - CRITICAL - cura.CrashHandler.__init__ [65]: AttributeError: 'NoneType' object has no attribute 'width'
2017-11-24 11:40:41,288 - DEBUG - AutoSave.AutoSave._onTimeout [48]: Autosaving preferences, instances and profiles

alekseisasin, 2017-11-24:
From the last log, part of error is related to CrashHandler, now it should not happen.

sfornengo, 2017-11-24:
I just waited for the ppa to be updated...
Cura interface opens now but cura crashes when I try to open a file.

franpoli, 2017-11-26:
I also have this error "Fatal Python error: Segmentation fault" on Fedora after the upgrade to version 27.
[cura_3.0.3.log](https://github.com/Ultimaker/Cura/files/1504142/cura_3.0.3.log)

alekseisasin, 2017-11-27:
One of the issues can be related to that Cura could not update the settings from the previous version. To check it you go to the home directory ( on Windows  C:\Users\<your username>\AppData\) and remove the cura folder and start Cura again. Before deleting please copy the C:\Users\<your username>\AppData\cura folder to another place. It will help us to find the reason of crash if the Cura works proparly.

sfornengo, 2017-11-27:
on ubuntu I already tried to remove cura settings: before cura crashed before opening interface, after removing settings the interface opens but crashed when trying to open a model or select a printer

franpoli, 2017-11-27:
Thanks @alekseisasin for the assistance. I have been removing all local files (.cache/cura/ .config/cura/ .local/share/cura/), it did not help. Then I looked back at issue [2558](https://github.com/Ultimaker/Cura/issues/2558) and the suggestion to install libglvnd-devel solved my issue on Fedora 27. @sfornengo, you may try this solution.

sfornengo, 2017-11-28:
Thanks @franpoli for the suggestion but there is no package libglvnd under my ubuntu 16.04, even with the cura master ppa. Did you compile it from source ?
Anyway I suspect it is the right track because I can see gl problem in all cura.log after crash:
2017-11-27 16:14:27,314 - DEBUG - UM.View.GL.ShaderProgram.load [55]: Loading shader file [/usr/share/uranium/resources/shaders/default.shader]...
2017-11-27 16:14:27,315 - DEBUG - UM.View.GL.ShaderProgram.load [55]: Loading shader file [/usr/share/cura/resources/shaders/grid.shader]...

ChrisRiddell, 2017-12-02:
After having alot of issues trying to run Cura from source thought I wanted to try the PPA and hit the same issue as you I think

it loads the machines but when loading the interface it crashes with the following in the Report a Problem dialog for Ubuntu 

"cura crashed with SIGSEGV in QWindow::geometry()"

[cura.log](https://github.com/Ultimaker/Cura/files/1523694/cura.log)

ChrisRiddell, 2017-12-02:
Okay I installed Ubuntu 16.04 LTS and used the PPA and it almost works in it opens you accept the agreement but crashes once you pick a printer eg Ultimaker 2+ 

GDB gives the following 

> 2017-12-03 09:14:26,432 - WARNING - UM.Qt.QtApplication.__onQmlWarning [270]: file:///usr/share/cura/resources/qml/SidebarSimple.qml:179: TypeError: Cannot read property of null
2017-12-03 09:14:26,433 - WARNING - UM.Qt.QtApplication.__onQmlWarning [270]: file:///usr/share/cura/resources/qml/SidebarSimple.qml:180: TypeError: Cannot read property of null
2017-12-03 09:14:26,434 - WARNING - UM.Qt.QtApplication.__onQmlWarning [270]: file:///usr/share/cura/resources/qml/SidebarSimple.qml:179: TypeError: Cannot read property of null
2017-12-03 09:14:26,434 - WARNING - UM.Qt.QtApplication.__onQmlWarning [270]: file:///usr/share/cura/resources/qml/SidebarSimple.qml:180: TypeError: Cannot read property of null
2017-12-03 09:14:26,441 - WARNING - UM.Qt.QtApplication.__onQmlWarning [270]: file:///usr/share/cura/resources/qml/SidebarSimple.qml:242: TypeError: Cannot read property of null
2017-12-03 09:14:26,459 - WARNING - UM.Qt.QtApplication.__onQmlWarning [270]: file:///usr/share/cura/resources/qml/SidebarSimple.qml:242: TypeError: Cannot read property of null
2017-12-03 09:14:26,492 - WARNING - UM.Qt.QtApplication.__onQmlWarning [270]: file:///usr/share/cura/resources/qml/SidebarSimple.qml:242: TypeError: Cannot read property of null
2017-12-03 09:14:26,879 - DEBUG - UM.Mesh.MeshData.approximateConvexHull [388]: approximateConvexHull(target_count=1024) Calculating 3D convex hull took 0.009527444839477539 seconds. 265 input vertices. 265 output vertices.
2017-12-03 09:14:26,884 - DEBUG - UM.View.GL.ShaderProgram.load [55]: Loading shader file [/usr/share/uranium/resources/shaders/default.shader]...
2017-12-03 09:14:26,886 - DEBUG - UM.View.GL.ShaderProgram.load [55]: Loading shader file [/usr/share/cura/resources/shaders/grid.shader]...
[New Thread 0x7fffca925700 (LWP 3170)]
[New Thread 0x7fff67fff700 (LWP 3171)]
Thread 1 "python3" received signal SIGSEGV, Segmentation fault.
0x00007ffff0c8e631 in QMetaObject::activate(QObject*, int, int, void**) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5

[cura.log](https://github.com/Ultimaker/Cura/files/1524325/cura.log)

sfornengo, 2017-12-11:
Could someone confirm that installing libglvnd solve this issue ?

thopiekar, 2017-12-11:
Hey @LipuFei, thank you for linking this issue to my packaging files.
Will see whether the libglvnd is installed here and if not whether it fixes the faulty starts.

thopiekar, 2017-12-11:
Installed libglvnd and all it's libraries, but that made Cura not working and even made my KDE desktop crashing. At least in version 0.1.0 it is not really useful on Ubuntu "Artful".

sfornengo, 2017-12-13:
Hey @thopiekar, thank you for trying that.
I am using ubuntu 16.04 with gnome desktop, how I could help to find the problem ?
The AppImage is working

lolorc, 2017-12-13:
been compiling & packaging cura myself on ubuntu 17.04 I also have this issue.
there's no libglvnd package in 17.04, so there's no -dev either.

thopiekar, 2018-02-11:
Cura 3.2 is built for Ubuntu Bionic and available in the stable PPA.

closed 2018-02-11 by thopiekar