Ultimaker/Cura #2832: Travel acceleration randomly ignored for initial calibration

14422, 2017-11-22:
Cura version:  3.0.4
Printer: Prusa i3 MK2
Printer drivers: 1.9.2
Printer firmware: 3.1.0
Octoprint: 1.3.5

Sometimes the travel acceleration is ignored during the initial calibration phase (which I believe uses "Start code" from my "Machine settings") and thus leads to quick instant movements thus being quite noisy in my case. I can't seem to find when it actually happens as sometimes even after rebooting my printer and reconnecting OctoPrint doesn't solve this.

I have set my printing and travel acceleration to 100mm/s from my regular print settings as I want my printer to be silent and I'm sending the G-code through WLAN connected OctoPrint  within Raspberry Pi 3 if that matters anything.

Is this a bug or are the acceleration settings even supposed to affect the movement speed during calibration?

14422, 2017-11-27:
Update: Travel acceleration during initial calibration seems to always work when I cancel a print through Octoprint and normally start a new print from Cura.

Ghostkeeper, 2017-12-11:
I don't think that your initial calibration phase is handled through Cura. Cura has no g-code for it as far as I know.

Maybe Cura leaves the acceleration settings at nonstandard values at the end of the print, and when you cancel you just got lucky?

ianpaschal, 2018-03-26:
Closing due to age and lack of info.

closed 2018-03-26 by ianpaschal