Ultimaker/Cura #2837: Cura 3.1B

Thisismydigitalself, 2017-11-24:
Win10/64
[image]

Thisismydigitalself, 2017-11-24:
[3.1B CrashLog.txt](https://github.com/Ultimaker/Cura/files/1501709/3.1B.CrashLog.txt)

alekseisasin, 2017-11-24:
Probably the problem is not related to your OpenGLVersion, from the log file I see the error occurs while initializing containter (it other word printer settings). 
I assume you have used custom printers in your previous cura version, if yes then can you provide additional information about them. To do this you can run old Cura load any model and save as project. (and share in github) The saved project will contain information of the used printer. This information might help us to understand better the issue.

Appesteijn, 2017-11-24:
Also reported at the forum: https://ultimaker.com/en/community/52135-ultimaker-cura-31-beta?page=1#reply-193559

Thisismydigitalself, 2017-11-24:
What should I be practicing to get this fixed?
Any file i should replace to get going?

fieldOfView, 2017-11-24:
Zip up your configuration folder (%APPDATA%\cura) and post it somewhere where it can be downloaded. Something is going wrong while loading your configuration, and we don't know what (yet).

The way to get it going would be to clear out the configuration folder (but leave the folder itself in place), but that will remove all your profiles so it should not be considered a solution.

Thisismydigitalself, 2017-11-24:
Unbelievable. 
Went into user folder, deleted my printer's *.inst.cfg file and Cura 3.1 is now up and running but without my printer which i will try to set again from scratch.

Thisismydigitalself, 2017-11-24:
Ok. this file is defiantly the culprit. 
[BW_current_settings.inst.zip](https://github.com/Ultimaker/Cura/files/1502396/BW_current_settings.inst.zip)

Thisismydigitalself, 2017-11-24:
I hope my findings will help fixing this bug.

[image]

Drayson80, 2017-11-25:
Same error happens on my side - have also a custom printer/derivate of UMO printer profile...

alekseisasin, 2017-11-27:
@Thisismydigitalself Can you upload your file again? The file which you uploaded is corrupted and cannot be opened.

fieldOfView, 2017-11-27:
@alekseisasin, the file @Thisismydigitalself is not actually a zip, but he renamed it to be a zip so he could attach it here.

@Thisismydigitalself, the file does not give is much information to go on, without the context of the rest of your configuration folder. Could you upload a zipped copy of your whole (not working) configuration?

ChrisTerBeke, 2017-11-27:
Actually the folder that represents the 3.0.x state would be quite useful as well so we can check the upgrading procedure between the versions for your specific configuration. The 3.1 folder might already have corrupted files which would be harder to check.

ChrisTerBeke, 2017-11-27:
(also we did a lot of fixes already, so bluntly loading your 3.1 config into our latest 3.1 local builds might cause a lot of other issues that we'll never encounter in real life).

LipuFei, 2017-12-13:
This is the same issue as https://github.com/Ultimaker/Cura/issues/2837

ChrisTerBeke, 2017-12-13:
#2837 is this issue indeed :P

Ghostkeeper, 2018-01-01:
So I believe the currently accepted answer to this one is: The upgrade from 3.0 to 3.1 left the folder in an invalid state. It's fixed in 3.2 but users of 3.1 will have to do a factory reset.

Or am I mixing up things now?

closed 2018-03-02 by ChrisTerBeke