Ultimaker/Cura #2839: Changelog: Wrong info about SolidWorks macro

thopiekar, 2017-11-24:
This is simply not true:
```
Add an "Export to Cura" button in SOLIDWORKS
SOLIDWORKS plugin can now be installed using an automatic installer.
```

foreachthing, 2017-11-27:
Yes ... this irritated me too - I was looking all over SolidWorks to find this ribbon/toolbar or export-button! It's on **import** plugin in Cura to **open** SolidWorks files.

ChrisTerBeke, 2017-11-27:
There's a macro available with installation instructions in the plugin repository. But I have to agree the change log description is incorrect.

foreachthing, 2017-11-27:
I've found that macro to be at `%APPDATA%\cura\3.0\plugins\CuraSolidWorksPlugin\CuraSolidWorksPlugin\macro\Export_to_Cura_2.x.swp`. Which was not documented either....

Ghostkeeper, 2017-12-05:
I've retroactively corrected the change log of 3.0 (it will appear correct in the 3.2 release).

As for the path to the SolidWorks macro, I think we'll have to re-make that video because indeed it should always save the macro to AppData, not `C:\workspace` (which was probably the location on @LipuFei's work laptop while running Cura from source).

closed 2017-12-05 by Ghostkeeper

thopiekar, 2017-12-06:
Officially said: The macro is already dead and not going to be maintained anymore. Also from my side there is no interest to continue any work on it.
That's why the video is not going to be updated by Ultimaker (as they created it). I personally have other priorities on the plugin at the moment and actually never cared much about the macro to be honest.

commit aa727df5 (2017-12-05): Correct description of SOLIDWORKS macro

The original description was factually incorrect.

Fixes #2839.