Ultimaker/Cura #2840: Collecting Data Prompt shown although disabled

thesec, 2017-11-24:
I disabled the preference to "Send (anonymous) print information".
Yet I am nagged with the prompt "Collecting Data ... You can disable this ... Dismiss" every single time I start Cura 3.0.4 :/

Is it another setting that I am missing? Please fix, if not.

ChrisTerBeke, 2017-11-26:
You'll have to click 'dismiss' instead of the 'x' in the top right corner.

thesec, 2017-11-27:
The legal requirements would need the dialog to be modal then plus they'd require the hint where and how to opt out.

When the according preference is unset the dialog must no longer appear. Or you could rename 'preferences' to 'afterrences'? ;D

Either way ... it is imho a bug.

closed 2017-12-05 by Ghostkeeper

Ghostkeeper, 2017-12-05:
I've made a fix that it doesn't display the message when the user disabled the setting without clicking on the message.

Black6spdZ, 2018-12-11:
This problem STILL plagues 3.6.0! There is NO dismiss button, only "ALLOW".. what kind of trickery is being pulled here?

maukcc, 2019-03-22:
Still showing up in 4.1

Ghostkeeper, 2019-03-25:
> This problem STILL plagues 3.6.0! There is NO dismiss button, only "ALLOW".. what kind of trickery is being pulled here?

We've originally put a "no, thanks" message there but had to hide it under an additional click for higher-ups because they seem to think that more data is better and want to include this dark pattern here to stimulate more people to send data.

A couple of things are weird about this message right now:
* The "Allow" button doesn't do anything except prevent the message from appearing again. It doesn't enable the analytics when you've previously disabled it.
* The "X" button only hides the message. It doesn't disallow analytics sending.

For 4.1 we're redesigning the first-use screens so that you don't get a million pop-ups any more about firmware upgrades being available, connecting stuff, change log, etc. but just one wizard. This wizard includes analytics, so hopefully it'll work a lot more nicely in this regard. It's not yet merged into Master though: https://github.com/Ultimaker/Cura/pull/5449

commit 13c7099e (2017-12-05): Don't show slice info message if the preference is set to not send it

If the preference was set to not send it but the user never clicked on the message, don't show the message.

Fixes #2840.