Ultimaker/Cura #2841: Move/Scale/... Dialogs not Closed

thesec, 2017-11-24:
Please close each of the dialogs when the edits are applied or allow to close these with e.g. a second click on the respective icon to free the screen space they unnecessarily occupy.

ChrisTerBeke, 2017-11-26:
You can click anywhere on the build plate where there's not an object and any tool menu will disappear.

thesec, 2017-11-27:
Yes, but the former menu appears again as soon as a model is selected. It should not.

ChrisTerBeke, 2017-11-27:
That's open for discussion, I guess the person that implemented this thought differently about it. I personally like that it opens the last used panel when I switch between selected models.

nallath, 2017-11-27:
Most of us have big monitors, so that's why screen real estate lost to not having to click an extra time.

thesec, 2017-11-27:
This is not about screen real estate but hiding behaviour from the user as is. Even if it was - that real estate is the user's and it is up to him to decide what to have on it.

With the tool being kept active atm the action to be executed and exptected is not visible and irritating.
E.g. open the scale tool, scale your object, close the tool by clicking anywhere. The tool and it's action are now hidden and no longer active. Now click on another model. Be sure not to move the mouse as that model will immediately be scaled ...

nallath, 2017-11-27:
They become hidden because no object is selected. You can't ever scale something if there is no object selected. 

We used to have that it would go back to the translate tool, but that got a lot of annoyed people because they seemed to expect that the old tool was still active (and then accidently ended up moving the object instead of scaling it).

thesec, 2017-11-27:
Hu? How would you accidentally apply anything if you explicitly activate and deactivate tools?!

Have no tool enabled by default and have them explicitly activated (and shown) and deactivated (and hidden) is the suggestion.
No one will end up accidentally applying anything when doing so and only when doing so.

Ghostkeeper, 2017-12-06:
What thesec is requesting doesn't have anything to do with the hiding functionality when no object is selected. He requests to be able to click the icon of the currently active tool to deactivate the tool, even if an object is selected.

This would make it easier to select a bunch of small objects and not get any tool handles in the way. Currently I tend to switch to Per-Model Settings when I want to do that (as it has no tool handles in the scene) but that's a bit weird.

Ghostkeeper, 2018-02-21:
Our project manager just removed this enhancement from our planning because it's not deemed important enough.

Ghostkeeper, 2018-02-21:
It's deferred until later notice. This usually means it's going to take a while unless we get lots of reports about it (in which case we tend to put it back in the planning as its priority is increased).

GregValiant, 2024-11-13:
A lot has changed in the interface since this was opened.  Can we close this?

closed 2024-12-10 by github-actions[bot]