Ultimaker/Cura #2843: Layer Count Indicator cut off

thesec, 2017-11-24:
Changing the layer height the new layer value (was 70 is now 60) is cut off on the left, see screenshot:
[image]

smartavionics, 2017-11-29:
Also doesn't cope with more than 999 layers!
[image]

Ghostkeeper, 2017-12-13:
We've had a few fixes related to this during the beta period of 3.1. Could you test again using the 3.1 stable version?

thesec, 2017-12-16:
Would love to but 3.1 messes up my settings from 3.0.4 so all materials, extruders and other are gone ... :/

thesec, 2017-12-21:
Confirmed done w/ me.

closed 2017-12-21 by thesec