Ultimaker/Cura #2844: "File"/"Save Project" opens "Save As" instead

thesec, 2017-11-24:
"File"/"Save Project" opens "Save As", please have it save the project file only.
A "Save As" dialog is expected with initial saving only, gcode exports should not account.

Ghostkeeper, 2017-12-01:
We expect there to always be a modal dialogue to select a file name. We don't yet keep track of the current project file, like with an asterix when the project has changed, etc.

Perhaps @mennoberg has an opinion?
If we change it to remember the project file and make "Save Project" behave like "Save" then we'd also need a "Save Project As" item.

zhou13, 2018-12-13:
In 3.6.0, I still only get "Save" in the Filename.  There is no "Save Project" and "Save Project As".  In addition, there is no "Ctrl-S" that I can conveniently use.

ipa64, 2021-05-18:
Now 4.9 still same weird behavior, sorry it's not an usual behavior, should have Save and Save as.