Ultimaker/Cura #2846: Change of settings not taking effect

ChrisRiddell, 2017-11-25:
Cura 3.1 Beta
Windows 10
Ultimaker 2+

When you change settings they are not taking effect eg: changing infill to 0 it still has infill or changing the wall thickness does not take effect ether.

[image]

Cura log.
[cura.log](https://github.com/Ultimaker/Cura/files/1502761/cura.log)

Thisismydigitalself, 2017-11-25:
I have the same issue with 3.1.
it looks like with 3.0 when we change a value it will auto refresh the settings.
with 3.1 however, you need to click on a different setting for the change to take effect.
not good.

i like where 3.1 is going. looking very promising, but i find it very slow compared to 3.0 and it still got issues. not complaining on a BETA version. just saying.

Appesteijn, 2017-11-27:
Devs: CURA-4627

ChrisTerBeke, 2017-11-27:
Will be fixed for the stable release (https://github.com/Ultimaker/Cura/commit/9e6704a1624535f04c7ee499f0d917e92ee431bf).

closed 2017-11-27 by ChrisTerBeke