Ultimaker/Cura #2850: Slicing Tolerance incorrect?

ChrisRiddell, 2017-11-26:
Cura 3.1 Beta
Windows 10
Ultimaker 2+

I have noticed when trying to use the Slicing Tolerance on some models eg the Tolerance Test from Makers Muse, it does not seem like its slicing correctly, Is this correct or a bug?

[image]
[image]

Vandrasc, 2017-11-27:
@BagelOrb, is this the expected behavior? It seems the inner walls are being not generated? This occurs with Normal 0.15 and Fast 0.2 quality when the slicing tolerance is set to Inclusive. Thank you!

BagelOrb, 2017-11-27:
It's not the inner walls, but inner holes which get removed.

Setting slicing mode to inclusive means that layers are generally expanded, so yeah I would expect.some holes to disappear.

I think this is expected behaviour.

Ghostkeeper, 2017-12-05:
I agree, this is expected behaviour. The confusion originates in the naming of the setting, I think.

"Inclusive" guarantees (bar printing failures and real-world effects like sagging) that the entire model is filled by the print. That is, the polygons that it finds after slicing are always at least as big as the print. Often bigger. In this case they are so much bigger that the space between the two walls completely disappears.

For reference, "Exclusive" does the opposite: It guarantees that the g-code is always contained within the model. "Middle" (the default behaviour) tries to average this such that the g-code best approximates the original model, sometimes going outside of the bounds of the model and sometimes not quite going all the way to the bounds.

closed 2017-12-05 by Ghostkeeper

ChrisRiddell, 2017-12-05:
Exclusive Gives the same effect as Inclusive.

I agree the name did confuse me as well as the tooltip description of what the difference between them

Ghostkeeper, 2017-12-05:
I still have the file loaded. For me, Exclusive properly opens up all those narrow rings. Just like with Middle, but Middle makes some of the rings a bit slimmer than Exclusive.