Ultimaker/Cura #2851: Cursor up/down keys: tilt the overview instead of running up/down slicing level; like turning left/right w/ cursor keys

RogerWeihrauch, 2017-11-26:
1st)
Thanks to all for integrating the new feature for printing simulation/show the way the slices are performed. It is amazing and will help me very much. Thanks
2nd)
I noticed using this feature the following possibility for enhancement: using the up/down cursor keys does no longer tilt the overview's cube; it runs up/down the slicing levels.
Whereas the left/right cursor keys turn the cube correctly.

What do you think?

Regards,
Roger

fieldOfView, 2017-11-26:
The up/down keys to step through the layers only works when the layer number box (or the "nubs" on the slider) has the focus. Click anywhere in the viewport to remove the focus from these items, and you can rotate the view with the cursor keys.

smartavionics, 2017-11-26:
> The up/down keys to step through the layers only works when the layer number box (or the "nubs" on the slider) has the focus. Click anywhere in the viewport to remove the focus from these items, and you can rotate the view with the cursor keys.

With the focus on the bottom layer slider the up down/keys move the slider up/down (as expected). Clicking on the build plate does not drop the focus from the slider as you describe above. Even clicking on a model on the build plate to select it does not change the focus. This is on a Linux system using the master branch.

fieldOfView, 2017-11-26:
You are right. This used to be possible. Not sure if it was changed intentionally or unintentionally in Cura 3.1

nallath, 2017-11-27:
I'm guessing unintentionally

Ghostkeeper, 2017-12-05:
Looks like it is still losing focus properly, but the keys still work on the layer scrolling rather than the camera even if the focus is not in the box.

The horizontal scrolling (per-line) seems to be working correctly, only when the focus is on the scroll bar.

diegopradogesto, 2018-01-05:
The UP/DOWN arrows move layers up and down because the layer number box takes the focus. In the line slider (horizontal) there isn't a box, so it's just taking the focus when clicking in the dot (it becomes blue). Then you can use the LEFT/RIGHT arrow to switch lines.
The right behavior for the layer slider should be the same. It's something that needs to be fixed.

Ghostkeeper, 2018-02-21:
Our project manager removed this issue from our planning because it was not deemed important enough.

closed 2023-06-24 by github-actions[bot]