Ultimaker/Cura #2852: Wrong Open Setting Indicator

thesec, 2017-11-26:
Keep one or more settings open, load a new project ... the previously open setting indicators (arrow left/closed and arrow bottom/open) are now in wrong order for these, see screenshot.
[image]

In addition, the characters of the open settings' heading get some funny alignment ;))

Ghostkeeper, 2017-12-01:
How can we reproduce this? It doesn't happen when I have some settings visible and then click on New Project.
Does it restore itself if you click twice on the category?

The arrows indicate whether the category was folded out. I think what might happen is that you loaded a project file which had some setting in the Build Plate Adhesion that is not applicable to your current set-up. For instance, it changed the raft air gap or something but you don't have raft enabled right now.

thesec, 2017-12-03:
Imo it happens when you open a project with the same settings and have at least one category opened.
It does not restore when you click twice.

ChrisTerBeke, 2017-12-03:
I have seen this a few times indeed when loading a project file that has category open/close settings in it.

Ghostkeeper, 2017-12-15:
Perhaps a project file would help to reproduce this then.

thesec, 2018-03-03:
Won't fix? No update? Is it fixed?

Ghostkeeper, 2018-03-03:
I think that was a mistake. The bug was confirmed but it's hard to reproduce. If you find it again, could you save a project file and send it over?

thesec, 2018-03-03:
Sure, will drop you a project file with next occurance.

ChrisTerBeke, 2018-03-03:
I closed it because we weren't able to reproduce it and there was no activity for almost 3 months. Now that there is and you still have the issue, it's fine to be re-opened.

thesec, 2018-04-11:
Can no longer reproduce, am closing it and will reopen when I can / should it happen again.

closed 2018-04-11 by thesec