Ultimaker/Cura #2853: Feature Request: Brin only inside

Drayson80, 2017-11-26:
As we have the possibility to place "Brim only on outside", it would be great to have the opposite too...
Currently printing an RC plane, would make the life easier in glueing the parts together...

Ando120, 2017-11-27:
Or maybe width setting for inside brim. I need only small inside brim.

Ghostkeeper, 2017-12-04:
So if you load an ordinary model which has no holes in the bottom layer you just see no brim at all. I'm putting it up in our issue planner but we have to consider usability a bit, because it might be unexpected to a user that just selects "Brim" but accidentally enabled this feature that no brim appears.

Drayson80, 2017-12-04:
Of course, it depeds on the model. if it has a solid bottom, then it is useless... but if something like a tube or a single layer wall print is required, then it is better to have a brim on the inside than outside. (easier to remove via a deburring tool than on outside).
Example: I need good adhesion AND a good outer surface at a single perimeter wall print of an RC plane (3dlabprint). Without brim, the print has limited adhesion, with brim, I have issues at cleanup and gluing the parts together...

nickthetait, 2017-12-04:
Rather than a sub setting of brim, what if these were three different adhesion strategies? Such as:
Brim (interior)
Brim (exterior)
Brim (both)

Ando120, 2017-12-05:
Sub setting for internal brim width will give all options (external or internal with can be 0 )

Materials with big thermal expansions need large external and small internal brim.

Ghostkeeper, 2018-02-12:
Our project manager removed this enhancement from our planning because it was not considered important enough.

Feidias04, 2019-06-07:
"Brim only on inside" - I need this option badly now. Because I would like to use "Spiralize Outer Contour" so I cannot prepare model with my CAD.  When I import this model and I would like to use "Spiralize Outer Contour", this bottom part isn't printed. But I would like to use this part - brim on inside for gluing parts together

enVide, 2019-09-14:
WE ALL NEED IT.

retif, 2020-04-01:
i need an option to turn on brim from both sides (for thin parts)

Ghostkeeper, 2020-04-02:
The option to turn the brim on for both the outside and the inside already exists: [Brim Only on Outside](https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/platform_adhesion/brim_outside_only.md). There is no option to turn that on automatically for thin parts, because something being thin is kind of ambiguous. It's not a per-object setting either since the brim is shared between all objects within a mesh group.

hazedav, 2020-12-12:
Bumping this.  It definitely makes way more sense to be able to have inside-only, outside-only, both pulldown for brim instead of a checkbox for "outside only".  Wouldn't inside-only be just as important as outside-only depending on the model?  I'm printing some wall switch covers and am definitely in need of some inside-only right now.

gloomytree, 2021-04-18:
I'd like this, too. My usecase at the moment: I have a model where it is really easy to remove the brim from the inside. The outside features a complex geometry, where it is really tedious to remove the brim.

yet-another-average-joe, 2021-04-27:
Here's a typical example of a part that would benefit from a inside only brim:

[image]

Printing upside down (gear up) is not an option there, and a brim on outside would be a mess (80 teeth...)

Creating the inside brim at CAD level + concentric 1st layer is not the same as a brim.

TCandee, 2021-05-07:
I have a very similar situation as "yet-another-average-joe".  Difficult to clean the brim from the teeth.  Please enable this function.

aaronse, 2021-06-11:
+1 Given inside-outside and outside are options, hopefully this isn't too hard to implement.

LutzFassl, 2021-10-03:
I need this too! I am maxing out the print bed with a thin part. So I need brim for adhesion, but cannot print on the outside --> PROBLEM. Need Brim Inside Only! Please

bechhansen, 2021-11-07:
I have often needed this and today I need it again.

LukasMueller187, 2021-11-08:
I'd like this feature too.

psiberfunk, 2021-11-25:
Also need this!  @Ghostkeeper : maybe your project/product manager could reconsider.. I would imagine it would be relatively easy to implement if you've already got the outside only function...

davidlip-cura, 2022-03-25:
Add me to the list of people wanting inside only brim.  The argument to use CAD to add it can also apply to any brim or raft so if inside only is not important, then any brim could be considered not important.

Inside only would make cleanup of many parts I make so much easier.  On some parts I can leave the inside brim.

Fredyy90, 2022-09-30:
A vital feature that is still missing from cura. Any updates on an ETA?

1knueller, 2022-10-02:
hello! id like to print something round with a thin, rough shell. it has to look good on the outside. brim only on inside would be an amazing feature.

smartavionics, 2022-10-02:
It's trivial to implement and will be in my next release (4.20.6). Check https://github.com/smartavionics/Cura/releases now and again.

[image]

yet-another-average-joe, 2022-10-02:
@smartavionics : do you plan Arachnid integration for your branch in a near future ? (Arachnid was a game changer for what I do ; I can't use your branch anymore :(  )

smartavionics, 2022-10-02:
Sorry, no. That's not going to happen.

PS - please everyone, don't spam this issue with any further comments related to my builds, for that go to https://github.com/smartavionics/Cura and open or contribute to an issue or discussion there, thanks.

psiberfunk, 2022-10-02:
> Sorry, no. That's not going to happen.
> 
> PS - please everyone, don't spam this issue with any further comments related to my builds, for that go to https://github.com/smartavionics/Cura and open or contribute to an issue or discussion there, thanks.

Would you be able to submit a pull request to the Cura team to take it up though?   That's not related to your build exactly, more like.. if you've developed this, it would be great if you could ask them to bring it in upstream :).

aaronse, 2022-10-02:
@Someone familiar with Cura and CuraEngine that's interested in putting a pull request together:

https://github.com/smartavionics/Cura/commit/e7c965cb6aeba2067d999b561851196c35001238 

https://github.com/smartavionics/CuraEngine/commit/56d86b7eba658adb51267935a282dbfb20e27e0a

aaronse, 2022-10-03:
Noticed @BagelOrb submitted pull request with very significant rewrite/improvements to Brim https://github.com/Ultimaker/Cura/pull/11678 maybe brim_inside_only will appear in, or after, that merges.

MariMakes, 2022-11-02:
@aaronse is right, it's easier for us to implement this now the PR from @BagelOrb is merged. 

We've added a ticket to the backlog with the intent to enable this. 
For internal reference CURA-9838

Thanks for the report! 👍

aaronse, 2022-11-02:
Thank you the update and info @MariMakes

Couldn't access CURA-9838 details, guessing jira item details are accessible to just Cura folks.

Hopefully the dev task description references smartavionics's change lists that seem to have implemented this feature already in their fork https://github.com/Ultimaker/Cura/issues/2853#issuecomment-1264745306 .  Hopefully dev estimate and risk are low enough this'll turn up soon.

Cheers!

poldim, 2023-03-13:
It looks like this is still not implemented in 5.3.0.  Can anyone with access to Jira advise when this might be added?

[image]

MariMakes, 2023-03-14:
Hey @poldim,

I took a look for you. 
[The contribution we referred to earlier ](https://github.com/Ultimaker/Cura/pull/11678) has been merged in Cura 5.3 so it should be easier to implement this feature. 

I'll flag it to the team again to increase the priority👍

aaronse, 2023-03-14:
Hello @MariMakes, cheers for the update!

Looks like this was first requested on [on Nov 26, 2017](https://github.com/Ultimaker/Cura/issues/2853#issue-276833447), which probably seems like a lifetime ago for all of us.  Is there something we can do to motivate the team to get this done?  Where are your Devs/Office located?  

If they're in the Seattle area I promise to drop off a 6-pack when this is done.  Maybe someone will offer the same if you're in a diff city, e.g. NYC.

DP-M, 2024-01-09:
It would sure be great to see this. Add me to the long list of people who have a thin walled part, which needs to have a clean outside. I'm using 5.6 and the option is not available on this release.

rff255, 2024-01-22:
I second this request. I have the very same need, for a thin tool with small marks on the outside:
[image]

ppriest, 2024-02-05:
I just encountered a case where this would have been useful - an objects sloping up from the outside, which an external brim usually makes even more visible when removed, but a hollow inside.
For the hopefully small amount of development and maintenance effort required (given inside brim is already supported as path of the !'Brim Only on the Outside', I reckon this is a win...

aaronse, 2024-02-05:
Happy New Year Everyone!

@MariMakes, due to inflation, and ludicrous interest rates, I will raise my bounty from 6 beers, to 7 beers.

There's at least one code fix mentioned in this topic already, and the offer of beer.  Befuddled as to why https://ultimaker.atlassian.net/browse/CURA-9838 is not resolved.  

Would be nice if bug database for open source project was openly shared, read access at least so community would have more context about what's happening. The context risks more questions, but it could also inspire more contributions too.

Ideally Customers could +1 up vote feature requests like this here, or on UserVoice Forum or something similar.

nallath, 2024-04-05:
Implemented in 5.7 :D

closed 2024-04-05 by nallath