Ultimaker/Cura #2855: BUG: post processing not applied when printing over network

UltiArjan, 2017-11-26:
firmware 4.0.1.20171023
cura 3.1 beta (but did not validate older versions)
pc: win 10
connection: wired network

Found what I think is a bug, no idea if it's specific for 3.1.

When using a post-processing g-code modifier, In my case "pause at height" on the UM3, it will not be applied when printing over the network directly from cura. If however you print via USB, or even if you first save the file somewhere and than print over network, the change will be applied.

fieldOfView, 2017-11-26:
Good catch. This is a regression introduced with Cura Connect. Should be fixed with the PR referenced above.

closed 2017-12-01 by ChrisTerBeke

ChrisTerBeke, 2018-01-17:
For reference: https://community.ultimaker.com/topic/21065-um3-cura-304-pause-on-height-not-work/.