Ultimaker/Cura #2858: Could we expose nozzle diameter in the GUI?

filipgoc, 2017-11-26:
I just wonder, are there many particular reasons why we make it impossible for the user to edit the nozzle diameter? 

It's currently only possible (user wise) by adding a Custom FDM printer. 

Now, what if a user likes to experiment (they do) and simply change the nozzle to something the manufacturer has not anticipated? Yet the nozzle size is the only difference. It would still be nice to get all the other defaults from the machine like bed shape and size, start and end gcode... and simply modify the existing profiles to get the new nozzle working. 

All other slicers I know have nozzle size as a user-editable setting. Why not Cura? 

Thanks for breaking this down to me.

ChrisTerBeke, 2017-11-26:
The nozzle diameter of a printer is part of a settings container we call a 'variant'. For example the Ultimaker 3 currently has 4 variants: AA0.8, AA0.4, BB0.8 and BB0.5. Next to these variants we make quality profiles that fit each combination of variant and allowed material, giving the user the best result without having to change settings. Each of these qualities has a lot of settings involved in them, and changing the nozzle diameter would result in a worse print quality, possibly even below our own standards. To prevent this from happening, we've locked down the diameters for all Ultimaker printers. Some printers however doesn't have a fixed nozzle size from the manufacturer, so that's the ones we allow to be changed by the user. Custom FDM printer has this, but also some others (like the BQ WitBox). Basically every machine that does not have custom extruder definitions.

In short you could say that we want Cura to be easier for the user, which sometimes means we have to hide functionality that could potentially disrupt the print result. One could even argue when you change the nozzle size, you have a whole different machine and would need to build the profiles from scratch anyways (which can be done by creating a custom FDM printer). Which machine are you trying to change the nozzle diameter of?

filipgoc, 2017-11-26:
Thanks for a swift reply @ChrisTerBeke!

I understand variants and how to make them. I'm one of the guys behind IMADE3D JellyBOX, and the task of making these things happen usually comes to me.

We have a quick release extruder that supports users using 3rd party hotends and whatever nozzles they may desire.

I already have a few 'official' variants with provided printing presets across different materials, but there are many nozzle sizes that I'm not ready to 'officially' support to the extent of creating super duper printing profiles.

At the same time, I don't see why make it more difficult for my users to experiment with nozzle sizes beyond what I have already given them by making them go with custom FDM. This would also mean they could not easily use the official JellyBOX profiles as a starting point (or their user generated profiles since those do not travel across machine types). 

I could just go ahead and create all imaginable variants and just do not provide any profile for them, but I feel that would be clearly confusing for beginners - defeating the whole purpose of why Cura is currently designed this way. 

To summarize, I am looking for a way to both have official variants, and to allow user to more easily customize the hotends and relevant profiles. 

PS: Is there way way I could expose the nozzle diameter in the machine setting for a printer that does have variants? I looked at the code a bit more closely now, but I don't see how I could override this default behavior. 

While this would be a half-measure, it would be more flexible than what we have now.

fieldOfView, 2017-11-27:
The nozzle size setting is available in Machine Settings for all printers *except* those that have variants, for the above mentioned reasons. Other than modifying the Machine Settings there is not much you can do about it (in Machine Settings).

For 95% of all purposes, you can use the setting "Line Width" as an equivalent of setting the Nozzle Diameter.

If you really want a Nozzle Diameter setting visible, you can add a new setting in your machine definition, and set the Nozzle Diameter setting to inherit from that new setting. In your variants, you would *not* change the nozzle size, but your own new setting instead. The nozzle size would automagically derive from that. 

Note that this last option is not officially supported, and may stop working at any time. Using Line Width is a much better idea.

filipgoc, 2017-11-27:
Thanks! I think using the Line Width is an acceptable hack for those wishing to experiment. I'll stick to that for now lest I end up with a broken piece of code when I least expect it.

filipgoc, 2017-11-27:
PS: that being said, I appreciate that you mention how one may accomplish this with custom variables in the hierarchical structure. This will come handy at some point. 😸

closed 2017-11-27 by filipgoc

Ghostkeeper, 2017-12-05:
For context, this choice was originally made way before we had a Machine Settings dialogue. The choice was that all properties related to the physical machine or firmware printer are made invisible in order to prevent confusion with basic users. They could forget that they had a different build volume size set for their printer and then destroy their printer by going outside of their build volume. We later went back on this decision partially when the Machine Settings dialogue was introduced, but it still prevents confusion when you select a 0.4mm nozzle and forget that you had the nozzle diameter setting set to 0.6mm.

The thought was also that whoever is advanced enough to physically modify their printer or firmware should also be advanced enough to modify a JSON file of Cura.

filipgoc, 2017-12-07:
Good context. That's always appreciated. 

To expand on the context (maybe not the right place): 

I guess the split is in what kinds of differences people consider a variant and what they consider a different machine. This is also machine dependent of course. 

---

> The thought was also that whoever is advanced enough to physically modify their printer or firmware should also be advanced enough to modify a JSON file of Cura.

Not sure about this one unfortunately. 

1. Even reasonably hands-on, handy people  may be completely lost when it comes to computers, software, or coding. Not everyone is comfortable both in hardware and software world. There is also a slight age divide on this one since computers are still a relatively new thing.
1. The current process of adding a printer variant is... not as easy as modifying a JSON file, and definitely a step above something like uncommenting a line in a configuration file. 
    - For a printer that already has a couple of variants, the process involves modifications and creations of at least three kinds of files that are paired through a combinations of names and keys.  (You know way more than I about these things.) That's a pretty high level of computer comfort required. 
1. The point to which 'modifying' a printer is an advanced operation depends on the printer. With some printers it may be very very easy to use 3rd party hardware or use custom nozzles to the point that it is beginners stuff. 

---

But all in all I don't see this as a very big issue since experimental user made nozzle profiles CAN be made by simply using the extrusion width! (Which I hadn't realized.)