Ultimaker/Cura #2859: Support XY distance does not work sometimes

cangelis, 2017-11-27:
I could not figure out exact reason why it overlaps in xy direction with the model even if I set xy spacings properly -I think-. This makes the support structure impossible to remove. See the screenshot to see the situation:

Support xy distance: 1.2
Minimum Support X/Y Distance: 1.2
Support Offset: 0.8

[image]

Version: Cura 3.0.4 on macOS Sierra

Please see the [link](https://www.dropbox.com/s/k1wlfpxh0a54hii/CFP_model_fixed.3mf?dl=0) for the cura project.

Thank you.

Vandrasc, 2017-11-27:
Devs, related to CURA-4607? ( @Ghostkeeper )

cangelis, 2017-11-28:
Is there any hack to prevent this, or any plan for a bugfix?

Vandrasc, 2017-11-29:
This will sound silly but can you change the position of the model a bit? In my case, when I rotate or change the position of the model the support gets correctly generated. You can also try to switch your Support Distance Priority to X/Y overrides Z, I hope this will do the trick. 

@ChrisTerBeke could you please have a look? Is 4607 related? Thank you!

cangelis, 2017-11-29:
I think I discovered the issue and seems xy-spacing does not work correctly at all. Please see the screenshot for the explanation.

[image]

Ghostkeeper, 2017-12-13:
CURA-4607 might be related, yes. (For those unable to read our Jira issue board: CURA-4607 reports a bug that the support X/Y distance seems to be inconsistent depending on the direction of the slope of overhang.)

The behaviour Cangelis showed in his last screenshot looks correct though. It's simply a matter of the priority of X/Y distance vs. Z distance, determined by the Support Distance Priority setting and the Minimum X/Y distance setting (if Z overrides X/Y).

ianpaschal, 2018-03-09:
@Ghostkeeper Is this fixed in the newer versions? I was messing around with these settings for my current project (as you know) and they seemed to work properly. @cangelis are you still experiencing this in the current version of Cura?

Ghostkeeper, 2018-04-04:
CURA-4607 was never fixed because we couldn't find the root cause and it would only happen in edge cases.

ianpaschal, 2018-05-28:
The behavior OP experiences seems to be the expected behavior (support x/y distance is only for x/y per layer so another layer can absolutely overlap. Nothing weird there. Since we haven't heard back for more than 2 months, it's safe to close.

closed 2018-05-28 by ianpaschal