Ultimaker/Cura #2860: Cura 3.04, fresh install - Text flickering and dissapearing

AdamS-DE, 2017-11-27:
HI,

I've just set up Cura on my workstation PC for our newly purchased Ultimaker 3 Extended.
Cura installed okay, without any errors, but when I open it the text on the screen (sometimes not all) dissapears if I stop moving the mouse. It usually reappears when I move the mouse, but trying to access any of the menus or buttons is almost impossible.
I'm not sure if this is an OpenGL error, but I have used other OpenGL software without issue.
The PC is running Windows 7 64-bit, and my graphics card is a ATI Firepro V5900.

Thanks,

Adam

[image]

ChrisTerBeke, 2017-11-27:
Hi. Do you have the latest version of your graphics drivers installed? We've seen rendering glitches being reported before and usually a driver update fixes it.

AdamS-DE, 2017-11-27:
It looks like my IT department haven't been keeping this PC up to date! Graphics drivers appear to be the cause!
Thanks.
Ads

closed 2017-11-27 by AdamS-DE