Ultimaker/Cura #2863: Cura opens Safari

LizardM4, 2017-11-27:
Cura 3.0.4
OSX 10.13.1 (17B48)

After a few minutes Cura is opened for the first time in the day, Safari starts on OSX. I use Chrome as default browser, so Safari stays put usually. No window is opened, just the app starts up and stays in the dock. I did several tests because I was not sure which program was the cause, but Safari does not open by itself unless Cura is open. This was happening also with the previous version of Cura.

ChrisTerBeke, 2017-11-27:
Which plugins do you have installed? I can't find anything that remotely relates to this in the Cura main codebase.

LizardM4, 2017-11-27:
I just downloaded the new 3.1.0 beta, and I saw Safari popping up under my own eyes after starting Cura. Before posting I've been reviewing every single service I have running on my Mac. If I have the chance of trying a clean install or a new user, I'll do it.

I also checked the `.app` bundle for any occurrence of the "popularmechanics" string, as well as this repo, but I found nothing.

I have no Safari plugin; I had UBlock Origin and 1Password, but I removed them nonetheless.

In Cura I did not add nor disable any plugin. Here is the list of the ones it shipped with:

 - 3MF Reader
 - 3MF Writer
 - Auto Save
 - Camera Tool (mandatory)
 - Changelog
 - Console Logger
 - Cura Profile Reader
 - Cura Profile Writer
 - CuraEngine Backend (mandatory)
 - File Logger (mandatory)
 - Firmware Update Checker
 - G-code Reader
 - GCode Profile Reader
 - GCode Writer (mandatory)
 - Image Reader
 - Legacy Cura Profile Reader
 - Local File Output Device (mandatory)
 - Machine Settings action
 - Material Profiles (mandatory)
 - Mirror Tool
 - Move Tool (mandatory)
 - OctoPrint connection
 - Per Model Settings Tool
 - Plugin Browser (mandatory)
 - Post Processing
 - Removable Drive Output Device Plugin
 - Rotate Tool
 - STL Reader (mandatory)
 - STL Writer
 - Scale Tool
 - Selection Tool (mandatory)
 - Simple View
 - Simulation View (mandatory)
 - Slice Info
 - Solid View (mandatory)
 - UM3 Network Connection
 - USB Printing
 - Ultimaker machine actions
 - Update Checker
 - UserAgreement
 - Version Upgrade 2.1 to 2.2
 - Version Upgrade 2.2 to 2.4
 - Version Upgrade 2.5 to 2.6
 - Version Upgrade 2.6 to 2.7
 - Version Upgrade 2.7 to 3.0
 - Version Upgrade 3.0 to 3.1
 - Wavefront OBJ Reader
 - Wavefront OBJ Writer
 - X-Ray View
 - X3D Reader

ChrisTerBeke, 2017-11-27:
I've never seen Cura do this and I have at least these plugins. Also given that there is no code in Cura that could trigger opening Safari on this specific website I don't think Cura is responsible for this behaviour. There's 4 places where Cura opens a browser (https://github.com/Ultimaker/Cura/search?utf8=✓&q=QDesktopServices.openUrl&type=) but it uses the default system browser there via QDesktopServices. Maybe it's one of these but there is some URL catching going on?

LizardM4, 2017-11-27:
I also did not think of Cura at the beginning; I got to it by excluding everything else (and I still cannot rule that's happening only on my system). Thanks for pointing those out, I will double check if `QDesktopServices` triggers Safari. I'm also trying to install OSX on a virtual machine to try it out on a clean install.

I will hopefully come back with more information.

LizardM4, 2017-11-28:
Ok it seems not to be Cura after all (I've been testing this on a virtual machine and I cannot make it happen).
My bad, sorry for the false alarm (but I really checked thoroughly, for weeks, and Cura is the only software I have installed since then).

closed 2017-11-28 by LizardM4

LizardM4, 2017-12-03:
It still *definitely* happens only *after* I have launched Cura at least once during the day. But it doesn't happen on High Sierra on my virtual machine. So it's something related to the environment in which Cura runs.

Ghostkeeper, 2017-12-15:
Cura would open Safari if you accidentally click one of the web links in Cura? Such as the material compatibility link. But then you'd expect a window to pop up with the web page you requested.

LizardM4, 2017-12-15:
No clicking involved. It is sufficient for Cura to be running. I just need to open the app and after a while a Safari icon pops in in the dock. It doesn’t try to browse any website (as far as Little Snitch can tell) and it doesn’t open any window.

LizardM4, 2018-01-21:
New OSX install, from scratch, did not use TimeMachine. Cura 3.1.0, everything is default, and I still observe this behavior. No interaction whatsoever with the program, and yet Safari launches itself. At this point it must be a Safari thing perhaps? Just why does it pop out with Cura, I can't figure out.

Ghostkeeper, 2018-02-08:
Hmm, does it happen at a certain point in the application, such as when you start a slice?

When you start a slice, Cura opens a local socket connection. Maybe Safari is doing something with that...? Though opening a local socket is quite common on Unix systems, so I'd expect other applications to do this too.

LizardM4, 2018-02-10:
No it is sufficient to open Cura. After a while, it pops up. I'm going to try to trace the Cura process, see if I get anything.

LizardM4, 2018-03-01:
Update: Cura 3.2.0, after quitting the app, Safari pops open. I quit it, it reopens. I quit it again (does not come back).
After 60 seconds, I start the newly downloaded Cura 3.2.1, first run, Safari pops open the moment Cura displays the changelog.

I'm clueless.

Ghostkeeper, 2018-03-26:
Me too, honestly.

Maybe it's worth looking at which file types are associated with Safari in your operating system?