Ultimaker/Cura #2866: Problem opening machine settings

gamebooya, 2017-11-28:
Version: 2.6.0-BETA
Platform: Darwin-17.2.0-x86_64-i386-64bit
Qt: 5.7.0
PyQt: 5.7.1

Exception:
Traceback (most recent call last):
  File "/Users/ultimaker/build/2.6/build/inst/lib/python3.5/site-packages/cura/MachineAction.py", line 57, in reset
  File "/Applications/Cura.app/Contents/MacOS/plugins/plugins/MachineSettingsAction/MachineSettingsAction.py", line 73, in _reset
    self._backend.disableTimer()
AttributeError: 'NoneType' object has no attribute 'disableTimer'

-------------------------
Thread 0x00007000159bd000 (most recent call first):
  File "/Users/ultimaker/build/2.6/build/inst/lib/python3.5/site-packages/UM/Backend/Backend.py", line 150 in _storeStderrToLogThread
  File "/Users/ultimaker/build/env/lib/python3.5/threading.py", line 862 in run
  File "/Users/ultimaker/build/env/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007000154ba000 (most recent call first):
  File "/Users/ultimaker/build/2.6/build/inst/lib/python3.5/site-packages/UM/Backend/Backend.py", line 142 in _storeOutputToLogThread
  File "/Users/ultimaker/build/env/lib/python3.5/threading.py", line 862 in run
  File "/Users/ultimaker/build/env/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x0000700014f34000 (most recent call first):
  File "/Applications/Cura.app/Contents/MacOS/plugins/plugins/USBPrinting/USBPrinterOutputDeviceManager.py", line 87 in _updateThread
  File "/Users/ultimaker/build/env/lib/python3.5/threading.py", line 862 in run
  File "/Users/ultimaker/build/env/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x0000700014a31000 (most recent call first):
  File "/Users/ultimaker/build/env/lib/python3.5/threading.py", line 297 in wait
  File "/Users/ultimaker/build/env/lib/python3.5/site-packages/zeroconf.py", line 1569 in wait
  File "/Users/ultimaker/build/env/lib/python3.5/site-packages/zeroconf.py", line 1153 in run
  File "/Users/ultimaker/build/env/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x000070001452e000 (most recent call first):
  File "/Users/ultimaker/build/env/lib/python3.5/threading.py", line 297 in wait
  File "/Users/ultimaker/build/env/lib/python3.5/site-packages/zeroconf.py", line 1569 in wait
  File "/Users/ultimaker/build/env/lib/python3.5/site-packages/zeroconf.py", line 1017 in run
  File "/Users/ultimaker/build/env/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x000070001402b000 (most recent call first):
  File "/Users/ultimaker/build/env/lib/python3.5/site-packages/zeroconf.py", line 946 in run
  File "/Users/ultimaker/build/env/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x0000700013b28000 (most recent call first):
  File "/Users/ultimaker/build/env/lib/python3.5/threading.py", line 297 in wait
  File "/Users/ultimaker/build/env/lib/python3.5/site-packages/zeroconf.py", line 1569 in wait
  File "/Users/ultimaker/build/env/lib/python3.5/site-packages/zeroconf.py", line 1017 in run
  File "/Users/ultimaker/build/env/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x0000700013625000 (most recent call first):
  File "/Users/ultimaker/build/env/lib/python3.5/site-packages/zeroconf.py", line 946 in run
  File "/Users/ultimaker/build/env/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x0000700013122000 (most recent call first):
  File "/Applications/Cura.app/Contents/MacOS/plugins/plugins/RemovableDriveOutputDevice/RemovableDrivePlugin.py", line 58 in _updateThread
  File "/Users/ultimaker/build/env/lib/python3.5/threading.py", line 862 in run
  File "/Users/ultimaker/build/env/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x0000700012c1f000 (most recent call first):
  File "/Users/ultimaker/build/env/lib/python3.5/threading.py", line 293 in wait
[... 75 lines cut ...]

Thread 0x000070000fa01000 (most recent call first):
  File "/Users/ultimaker/build/env/lib/python3.5/site-packages/zeroconf.py", line 946 in run
  File "/Users/ultimaker/build/env/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/lib/python3.5/threading.py", line 882 in _bootstrap

Current thread 0x00007fffb5001340 (most recent call first):
  File "/Users/ultimaker/build/2.6/build/inst/lib/python3.5/site-packages/cura/CrashHandler.py", line 100 in show
  File "/Users/ultimaker/build/2.6/build/inst/bin/cura_app.py", line 44 in exceptHook
  File "/Users/ultimaker/build/2.6/build/inst/lib/python3.5/site-packages/cura/CuraApplication.py", line 659 in run
  File "/Users/ultimaker/build/2.6/build/inst/bin/cura_app.py", line 70 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load
  File "/Users/ultimaker/build/env/lib/python3.5/site-packages/cx_Freeze/initscripts/Console.py", line 21 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load
  File "/Users/ultimaker/build/env/lib/python3.5/site-packages/cx_Freeze/initscripts/__startup__.py", line 12 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load

fieldOfView, 2017-11-28:
May I ask why you are running a beta of Cura 2.6? Please try with a non-beta version.

ChrisTerBeke, 2017-11-28:
This should work properly in all newer versions of Cura, so like @fieldOfView suggests please use a newer version.

closed 2017-11-28 by ChrisTerBeke

gamebooya, 2017-11-28:
Thanks, I updated to the newest version of Cura 3.0.4 it works now. I had just updated my operating system and I think that was part of the problem.

fieldOfView, 2017-11-28:
No, this was a bug that was introduced in Cura 2.6 beta, but fixed for the final Cura 2.6 as far as I remember.