Ultimaker/Cura #2871: Layer view extremly slow Cura 3.0.4 win64

rekom, 2017-11-28:
I´m using cura 3.0.4 on win10 64bits with an AMD APU computer.
When in layer view, rotating the model is extremly slow. This is fixed with the option "force layer compatibility mode". 
In the windows monitor, I can see that the iGPU (there's no discrete gpu) is used at 100% in 3d. 
When i use an intel haswell iGPU, (also without discrete GPU) it runs ok.
Weirdly, in 3d benchmarks (3d Mark) AMD iGPU is faster than the intel one.

Could it be a driver problem?

ChrisTerBeke, 2017-11-28:
Do you have the latest video drivers for the discrete GPU installed?

rekom, 2017-11-28:
It´s the lasted drivers but i don´t have a discrete card, it´s just the iGPU rendering

rekom, 2017-11-28:
Just tested a luxmark benchmark in this case the AMD GPU is much more slower than the intel one...

fieldOfView, 2017-11-28:
This is the reason "force compatibility mode" is there; some GPUs support the required functionality for the modern layer view, but only at a snail's pace. Sometimes newer drivers help, but for some GPUs the hardware is just too slow. Fortunately there's a partial fallback that is much faster.

Ghostkeeper, 2017-12-13:
I hope that compatibility mode is sufficient for you. I've also just created a pull request that should almost double the frame rate of layer view: https://github.com/Ultimaker/Cura/pull/2970 (but that's maybe for next release).

closed 2017-12-13 by Ghostkeeper

rekom, 2017-12-14:
@Ghostkeeper Same machine Cura 3.1 with last AMD drivers (but i believe it didn´t change anything the APU is too old to get fix anymore0) and it works really really well. It´s not 100 fps/s but i can really use the ´normal´mode!