Ultimaker/Cura #2873: Material Colour always yellow

kr15uk, 2017-11-28:
Cura 3.0.4
macOS 10.13.1 High Sierra
Custom machine
Custom material with set colour

In Layer View, Material Colour model is always yellow, doesn't change to set colour.

ChrisTerBeke, 2017-11-28:
I cannot reproduce this issue. Did you activate (select) the custom material after creating it?

kr15uk, 2017-11-29:
Yes I did.
It works in Solid View but not in Layer View (Material Colour)

ChrisTerBeke, 2017-11-29:
Ok. And I assume it re-sliced after selecting the custom material?

kr15uk, 2017-11-29:
Yep, every time when I change material it re-slices.

ChrisTerBeke, 2017-11-29:
Okay, I can reproduce this in 3.0.4. In the 3.1 beta it seems fixed for the layer view (we overhauled the layer view and extruder settings management), but the color indicator circles don't seem to update yet, so I'll create an internal ticket for those.

ChrisTerBeke, 2017-11-29:
Devs: see CURA-4641

ChrisTerBeke, 2017-11-29:
It was fixed in https://github.com/Ultimaker/Cura/commit/16e07fc461ad186038c6f8aa2cf55847593bd1ec. Please keep in mind that the layer view color won't update until after a re-slice. This is expected behaviour.

closed 2017-11-29 by ChrisTerBeke